RobotLocomotion/drake #6099: doxygen doesn't render HTML char entities properly

soonho-tri, 2017-05-15:
Example: http://drake.mit.edu/doxygen_cxx/group__solvers.html

It ignores (`&diams;`, :diamonds:) and (`&dagger;`, †) in https://github.com/RobotLocomotion/drake/blob/master/drake/solvers/mathematical_program.h#L76-L80.

[image]

It says it built the documentation using `doxygen-1.8.6`. We can't reproduce the problem with later versions such as `doxygen-1.8.13` (OSX) and `doxygen-1.8.11` (Ubuntu Xenial).

jamiesnape, 2017-05-15:
@BetsyMcPhail Switch the documentation builds to Xenial. You will need to install the oXygen XML Editor onto the Xenial image and remove it from the Trusty image.

BetsyMcPhail, 2017-05-29:
The documentation build has been switched from Trusty to Xenial. 

As a result, doxygen has also been updated to 1.8.11.

closed 2017-05-29 by BetsyMcPhail

soonho-tri, 2017-05-29:
Nice!

stonier, 2017-06-01:
@soonho-tri should the [Mixed-Integer Convex Optimization](http://drake.mit.edu/doxygen_cxx/group__solvers.html) table be empty?

jwnimmer-tri, 2017-06-01:
Yes https://github.com/RobotLocomotion/drake/blob/78a2bf6d50ca4c66fceebf30cb5982734033024e/drake/solvers/mathematical_program.h#L94-L118

soonho-tri, 2017-06-01:
Both of Mosek and Gurobi support mixed integer programming [[1](http://www.gurobi.com/resources/getting-started/mip-basics), [2](http://docs.mosek.com/7.0/tools/The_optimizers_for_mixed-integer_problem.html)]. I will ask @hongkai-dai for details and open a PR to update the doc.