RobotLocomotion/drake #6108: Simplify MosekSolver::Solve error semantics

EricCousineau-TRI, 2017-05-16:
(Incurring technical debt to simplify PR #6085.)

Some of the logic within `MosekSolver::Solve` could be reduced by:
* Using some sort of `ScopeExit` mechanism to (a) free the task, and (b) set the solver type.
* Throwing exceptions when there are infrastructure issues (such as when `MSK_appendvars` fails).

An attempt was made in this commit (but needs further revision, per Jeremy's comments in the original PR):
https://github.com/EricCousineau-TRI/drake/commit/1a3bd6f684a487e8ad8dd0515ac36107df495fad

jwnimmer-tri, 2019-03-26:
This has not been refactored yet, but the need does not seem to be acute.  @EricCousineau-TRI please either fix it now, or turn this issue into a TODO in the code so that we can close it and stop cluttering the issues database.

EricCousineau-TRI, 2019-03-27:
Ehhhh, looking back on it now, I don't think this addresses any major pain points. I don't think it's even worth a TODO for now.

Football!

(... because punting?)

closed 2019-03-27 by EricCousineau-TRI