RobotLocomotion/drake #6109: UnitInertia can be modified to violate its invariants

sherm1, 2017-05-16:
Currently UnitInertia derives from RotationalInertia, but the latter provides mutators that can't be permitted on a UnitInertia.

Thanks to @jwnimmer-tri for pointing this out in [this comment](https://github.com/RobotLocomotion/drake/pull/5300#issuecomment-301647269).

Likely the class hierarchy should have been something like:
```
RotationalInertia
    UnitInertia
    GeneralInertia
```
with the scary mutators moved to GeneralInertia. 

Possibly this can be addressed in PR #5945 which is currently in review.

amcastro-tri, 2017-05-16:
`GeneralInertia`? yuck! Hopefully we can find a better solution but if not maybe:

    RotationalInertiaBase (abstract)
      UnitInertia
      RotationalInertia

Also, given that getting a functional MBT is the highest priority for us now (i.e. to provide a full simulation of a Kuka arm with it), and given this would not change the visible (and most used) API for `RotationalInertia` I decided to lower this issue's priority.

jwnimmer-tri, 2017-05-16:
@amcastro-tri I think you're saying that `UnitInertia` is not widely used, so the fact that its broken is low priority?  I could go along with that.  Right now, only the `SpatialInertia` constructor meaningfully uses it, and I think that constructor could gain a fail-fast check that the argument is properly constrained.

sherm1, 2017-05-16:
The base class would not need to be abstract in this case, @amcastro-tri. It would be exactly the RotationalInertia class we have now, less the mutator methods. The other two classes would be data-free and final. That way there is no performance cost. I'm not sure that's a good idea though -- another possibility would be to eliminate the bad mutator methods altogether -- do we actually need them for anything?

jwnimmer-tri, 2017-05-16:
The mutator method that is both the most hazardous and yet the most important to keep intact is the default copy-assignment operator, `RotationalInertia::operator=(const RotationalInertia&)`.

sherm1, 2017-05-16:
The base class would have to suppress copy assignment, but could provide `RotationalInertia::operator=(const UnitInertia&)`. The looser derived class would have `GeneralInertia::operator=(const RotationalInertia&)` and copy assignment. Not pretty. Likely a UnitInertia that has-A RotationalInertia would be nicer.

sherm1, 2017-05-16:
I think this would work, enforce invariants, and be backwards compatible:
```c++
RotationalInertiaBase // data, most functionality, anything dangerous is protected, not virtual
    RotationalInertia // final, no data, adds missing mutators
    UnitInertia  // final, no data, adds implicit conversion to const RotationalInertia
```

In this case RotationalInertiaBase is not intended to appear anywhere, it is just there to collect the large amount of shared code. RotationalInertia serves as the generic inertia class, since it will accept either RotationalInertia objects or implicitly-converted UnitInertia objects.