RobotLocomotion/drake #6113: [WIP] Output port separation improvements

sherm1, 2017-05-16:
(please ignore for now -- this issue is still being assembled)

PR #6038 changed the output port API to allow separate per-port computation. Many good ideas came up in review that should be considered for follow-on improvements. This issue collects them:

- [ ] Output port error messages include the System name and type but only the port index. Add the port type to the message. (@SeanCurtis-TRI [comment](https://reviewable.io/reviews/robotlocomotion/drake/6038#-KjiJ2lYXOqgfrfQ1ndM)).
- [ ] Using "Alloc" for the member functions that return an object directly is misleading since we usually use that verb to indicate a heap allocation. Should pick another verb. (@amcastro-tri) @edrumwri suggested "Instantiate".

More ...

jwnimmer-tri, 2017-05-16:
Re: naming.  I agree that "Allocate" is not ideal; really, it's just wrapping a constructor, so just "Construct"?

jwnimmer-tri, 2021-11-11:
I'm guessing here, but I'd wager that the `PortEvalCast` refactoring fixed the only remaining open checklist item, so I think we can call this done.

closed 2021-11-11 by jwnimmer-tri