RobotLocomotion/drake #6116: Make Gurobi license checkout errors more obvious

EricCousineau-TRI, 2017-05-16:
(Related to #6108)
Presently, if Gurobi cannot check out a license, the solver returns NaNs and an unsuccessful status (`kInvalidInput`).
Instead, we can either `DRAKE_DEMAND` or `throw` a Gurobi-related exception, and include the output from `GRBgeterrormsg(env)`.

This will address the present TODO in `GurobiSolver::Solve(...)`.

I came across this when fixing #6110, where `//drake/multibody/dev:global_inverse_kinematics_reachable_test` failed and it was non-obvious that it was a licensing error.

\cc @hongkai-dai @naveenoid

naveenoid, 2017-05-16:
Is there no API for probing the license server? Solver returning NaNs could also indicate bad problem setup...

hongkai-dai, 2017-05-16:
> Solver returning NaNs could also indicate bad problem setup...

I think that can mean the problem is infeasible, but not necessary a bad problem.

naveenoid, 2017-05-16:
Sorry I meant returning NaNs + kInvalidInput - is that sufficient?

hongkai-dai, 2017-05-16:
Yes, exactly as @naveenoid  says, NaNs + kInvalidInput could mean the input is bad, but the license is there. For example, the program
min x
s.t 1 <= x <= 0
Will cause kInvalidInput + NaNs in Mosek.

How about adding a new type to the enum `SolutionType`, as `kInvalidLicense`?

EricCousineau-TRI, 2017-05-16:
Yup, I can see if there is a mechanism for probing the license server, or just check the error return code, and ensure that either (a) `SolutionType` to be accurate or (b) we provide the appropriate information about the exception.

TBH, I would prefer an exception w.r.t. license checking, as that should not be happening at all.
If we return `kInvalidLicense`, to inspect the failure of a GTest, would we have to look at the actual return value (which I believe would be printed as an integer), and then cross-reference that with the enum definition. Inside of an actual binary, we would have to do a similar test, or explicitly check the error type.

In contrast, with an exception (at least for licenses), the problem is much more clear, and there is no effective recovery mechanism (which is what I would expect if we were to just return the failure type.)

@hongkai-dai Are there situations where you think throwing an exception would cause an issue?

hongkai-dai, 2017-05-16:
Yes, throwing an exception sounds good.

EricCousineau-TRI, 2018-12-08:
After having let this sit for a while, this doesn't seem to be much of an issue. Going to close for now. Can re-open if need be.

closed 2018-12-08 by EricCousineau-TRI