RobotLocomotion/drake #6123: SNOPT problem

GuiyangXin, 2017-05-17:
Hello all, I reinstalled Drake following the guidance of http://drake.mit.edu/from_source.html instead of following oh-distro-private. So I have to install SNOPT separately. And I found this issue #3369. Following the advice from @samkhal , I applied a snopt7.lic file and the snoptmex file. Cloned his wrapper for snopt https://github.com/samkhal/snopt-matlab
But when I run runAtlasWalking.m, error occurred
```
>> runAtlasWalking

ans =

     1

Subscript indices must either be real positive integers or logicals.

Error in snopt (line 148)
F0 = userFG(x);

Error in NonlinearProgram/snopt (line 1442)
        [x_free,objval,exitflag,xmul,Fmul] = snopt(x0_free, ...

Error in NonlinearProgram/solve (line 985)
            [x,objval,exitflag,infeasible_constraint_name] = snopt(obj,x0);

Error in DynamicalSystem/resolveConstraints (line 317)
      [x,~,exitflag,infeasible_constraint_name] = solve(prog,x0);

Error in TimeSteppingRigidBodyManipulator/resolveConstraints (line 1022)
      [varargout{:}] = resolveConstraints(obj.manip,x0,varargin{:});

Error in Atlas (line 102)
        obj = obj.setInitialState(obj.resolveConstraints(zeros(obj.getNumStates(),1)));

Error in runAtlasWalking (line 26)
r =
Atlas(fullfile(getDrakePath,'examples','Atlas','urdf','atlas_minimal_contact.urdf'),robot_options); 
```
The function in snopt.m is defined as `function [x,F,inform,xmul,Fmul,xstate,Fstate,output] = snopt( x, xlow, xupp, xmul, xstate,
						  Flow, Fupp, Fmul, Fstate,
						  userfun, varargin );` . However, the invoking of this function in NonlinearProgram.m is ` [x_free,objval,exitflag,xmul,Fmul] = snopt(x0_free, 
          x_lb_free,x_ub_free, 
          lb,ub,
          'snoptUserfun',
          0,1,
          Avals,iAfun,jAvar,
          iGfun_free,jGvar_free);` . Obviously, they don't match well, which results in the error mentioned above. Fortunately, I found this issue #1488. However, after I changed line 1442 in NonlinearProgram.m as `[x_free,objval,exitflag,xmul,Fmul] = snopt(x0_free,
          x_lb_free,x_ub_free, 
          [],[],lb,ub,
          [],[],'snoptUserfun',
          0,1,
          Avals,iAfun,jAvar,
          iGfun_free,jGvar_free);`, new error happened as follows
```
runAtlasWalking

ans =

     1

Error using snoptmex
Wrong number of input arguments

Error in snopt (line 250)
[x,F,inform,xmul,Fmul,xstate,Fstate,itn,mjritn] = snoptmex( solveopt, x, ...

Error in NonlinearProgram/snopt (line 1442)
        [x_free,objval,exitflag,xmul,Fmul] = snopt(x0_free, ...

Error in NonlinearProgram/solve (line 985)
            [x,objval,exitflag,infeasible_constraint_name] = snopt(obj,x0);

Error in DynamicalSystem/resolveConstraints (line 317)
      [x,~,exitflag,infeasible_constraint_name] = solve(prog,x0);

Error in TimeSteppingRigidBodyManipulator/resolveConstraints (line 1022)
      [varargout{:}] = resolveConstraints(obj.manip,x0,varargin{:});

Error in Atlas (line 102)
        obj = obj.setInitialState(obj.resolveConstraints(zeros(obj.getNumStates(),1)));

Error in runAtlasWalking (line 26)
r =
Atlas(fullfile(getDrakePath,'examples','Atlas','urdf','atlas_minimal_contact.urdf'),robot_options); 
```
So, I don't know how to deal with this error. Anyone can help me, thank you.

hongkai-dai, 2017-05-18:
@samkhal have you seen this error before?

Could you add a break point at line 250 of snopt.m, read the input arguments, and check if these arguments match the API of snoptmex?

GuiyangXin, 2017-05-18:
Thanks guys, I think it is because I didn't install SNOPT correctly because I cannot run `>> runNPexamples  % to test the NLP solver (SNOPT)` successfully. I have switched to use SNOPT 7.2. So Atlas can walk very well. However, my previous problem about changing friction coefficient between feet and terrain still cannot be solved. I change line 65 as  `mu = 0*ones(nC,1);` in contactConstraints.m file. Atlas still can walk very well without any difference compared with when  `mu = ones(nC,1);`. And when running runAtlasBalancingPerturb.m, it didn't show any slippy feature as well.

hongkai-dai, 2017-05-19:
Drake uses a time stepping approach to simulate the robot. It calls `SolveLCP` function  in `TimeSteppingRigidBodyManipulator.m` in every step of simulation. It seems to me that the default option is to call the mex version of this function in https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/systems/plants/TimeSteppingRigidBodyManipulator.m#L288~L291. Unfortunately in this mex version of the function (solveLCPmex), the variable `mu` in https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/systems/plants/TimeSteppingRigidBodyManipulator.m#L269 is not used. There are two ways to fix this problem, either in the MATLAB code, or in the cpp code.

1. To comment out line 288-291 of TimeSteppingRigidBodyManipulator.m, and `solveLCP` will call the matlab version, which uses `mu` in https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/systems/plants/TimeSteppingRigidBodyManipulator.m#L648. 
2. Another way is to change the mex version of `solveLCPmex.cpp` in https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/systems/plants/solveLCPmex.cpp#L357, then recompile drake through cmake. And then you can run the simulation, without commenting out line 288-291 of `TimeSteppingRigidBodyManipulator`

The MATLAB version of `solveLCP` is slower in runtime than the mex version `solveLCPmex.cpp`. So I would recommend trying approach 2. You do need to re-compile drake every time you change a `cpp` file.

GuiyangXin, 2017-05-20:
Yes, your suggestion is right. Both methods are correct. After changing `solveLCP`, Atlas cannot walk any more because of slipping with mu=0.1. But my vision of making LittleDog slip on the ground when it is applied external force failed. I think it is due to my algorithm rather than Drake. So thank you all for your help. This issue has been solved and can be closed. Thanks again.

closed 2017-05-20 by hongkai-dai

Jerryxiaoyu, 2018-11-08:
@GuiyangXin I encountered the same problem with SNOPT. Should I install the old version, SNOPT 7.2 to fix it? But where can I download 7.2? I can't find it. 
Could send me a copy? Thanks

GuiyangXin, 2018-11-08:
SNOPT is not an open source software. Sorry.