RobotLocomotion/drake #6129: "Slow read"s when building with Bazel

m-chaturvedi, 2017-05-17:
For example:
https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-bazel-experimental-memcheck-valgrind-everything/7/console

at time `17:15:27` in the log above.

jwnimmer-tri, 2017-05-17:
Probably just due to things like #5900, exacerbated by CI being slower than local workstations.  We might be able to disable the warning, but I'm not sure what else useful we'd expect to do in this issue.

stonier, 2017-05-25:
@m-chaturvedi brain dump whatever you learn to this issue!

m-chaturvedi, 2017-05-25:
- This github issue seems related to the problem, https://github.com/bazelbuild/bazel/issues/1210 , #835 which it cites in particular.
- The flag `--local-resources` which can set the available IO (1.0 by default) maybe useful after the changes made in https://github.com/bazelbuild/bazel/issues/835
- When running on my machine I observed higher latency ( external HDD ) gives more number of "slow read"s.

jwnimmer-tri, 2017-06-19:
All issues must have owners.  I'm assigning one arbitrarily.  (Fix it up if I'm wrong.)

stonier, 2017-08-10:
From Jamie : can often happen when a new instance is spun up - the remote hard drive is pretty slow getting written to the first time and can trigger this problem. Instances get culled and spun up pretty frequently.

PR #6904 (2017-09-04): Add note about slow reads during Bazel builds to build cop playbook
Closes #6129.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6904)
commit 77bf8ecb (2017-09-04): Add note about slow reads during Bazel builds to build cop playbook

Closes #6129.
closed 2017-09-04 by jamiesnape