RobotLocomotion/drake #6134: Consider more efficient / non-redundant storage and computation for AutoDiff orders > 1

EricCousineau-TRI, 2017-05-18:
(This may have been discussed before, but just putting it in issue form.)
Eigen's AutoDiff formulation will consume redundant memory / calculations, as mentioned in the [Wikipedia article](https://en.wikipedia.org/wiki/Automatic_differentiation#High_order_and_many_variables).

Presently, it uses the algebra mentioned in the article which yields nth-order representations like this:
```
Order    Representation
-----    --------------
0        u
1       (u, du)
2     < (u, du), (du, ddu) >
3  [  < (u, du), (du, ddu) >,  < (du, ddu), (ddu, dddu) >  ]
```
which ultimately turns into Pascal's triangle, and the number of redundant variables (and computations) will be `n^2 - n - 1` (e.g., for 2nd order, we have two distinct instances of `du` that we have to put effort in to make symmetric, in `value().derivatives()` and `derivatives().value()`).

For a concrete reference, here is a quick-and-dirty example for nth-order AutoDiff types with some other comments:
https://github.com/EricCousineau-TRI/repro/blob/5ba6532/eigen_scratch/autodiff_nth_order.cc#L39

For linear operations, this duplication is not necessary, and we could easily leverage an expression of the form `[u, du, ddu, ...]`.
For nonlinear operations, the nested duplications are necessary to capture the inner product terms.

We could consider some workarounds (if we think there are ever performance bottlenecks for nth-order derivatives at some point in the future):

* (Semi-low-hanging fruit, Alejandro's suggestion) We could specialize 2nd order AutoDiffs for Hessians, since that may be as far as we go. It would be easy-ish to define the non-general algebra for 2nd order AutoDiffs of the form `(u, du, ddu)`.
* Linked from the Wikipedia article is a library which appears to compute arbitrary nth order derivatives:
https://github.com/zigasajovic/dCpp
    * The library seems to handle the scalar and multivariate case, but does not seem to handle multidimensional storage (which may be a good thing for a different form of sparsity?).
    * We could use the `tau` operations involving nth order flattened AutoDiffs, `(u, du, ddu, ....)`

I had tried out flattening the `AutoDiff` structure by changing the template to `AutoDiffStorage<Scalar, Derivative>`, such that you could define (pseudo-code):
```c++
using AutoDiffNScalar<N, Scalar> = AutoDiffStorage<Scalar, AutoDiffStorage<N - 1, Scalar>>
using AutoDiffNScalar<0, Scalar> = Scalar;
```
but that was, uh... before I realized the algebra was completely invalid for `N > 1`. \</embarrassment\>
https://github.com/EricCousineau-TRI/repro/blob/5ba6532/eigen_scratch/AutoDiffScalarMod.h

\cc @amcastro-tri @tkoolen @hongkai-dai

RussTedrake, 2017-05-18:
Not the same, but this is related: #2619

amcastro-tri, 2017-05-18:
Thanks for putting this in writing. 
@edrumwri actually wrote a beautiful pdf with very similar thoughts now I remember

hongkai-dai, 2019-03-26:
Close it for now as we don't have a use case

closed 2019-03-26 by hongkai-dai