RobotLocomotion/drake #6135: Parse a simple double pendulum with sdformat

amcastro-tri, 2017-05-18:
Recently merged PR #6028 introduces the `sdformat` parser into Drake (yai!).
As a proof of life, this PR introduces a [demo test](https://github.com/RobotLocomotion/drake/blob/master/drake/multibody/parsers/test/sdformat_test.cc) for a simple world containing a single link. How would we go about parsing a file with more than one link? take for instance this very simple example of a [double pendulum](https://github.com/RobotLocomotion/drake/blob/master/drake/multibody/benchmarks/acrobot/double_pendulum.urdf). 
It'd seem we'd need to make repeated calls to `Element::GetNextElement()` since `Element::GetElement()` seems to always return the first child (is that true? there is no doc either of these).

I think a very good next goal for integrating sdformat into Drake would be to succesfully parse [double pendulum](https://github.com/RobotLocomotion/drake/blob/master/drake/multibody/benchmarks/acrobot/double_pendulum.urdf) (actually it's sdf version).

amcastro-tri, 2017-05-18:
cc'ing @clalancette, @stonier, @sherm1

sherm1, 2017-05-18:
Maybe the next step should be to improve sdformat's documentation?

stonier, 2017-05-18:
Already bumped OSRC to watch this space and support it with documentation where you guys have questions.

stonier, 2017-05-26:
I asked the lads from OSRC to run with this and create a more complex introductory example illustrating how to parse it with the c++ api. 

We also talked about a c++ api and tool that could be used to validate them. Apparently there is one in gazebo that could/should be more logically in sdformat. 

@caguero

stonier, 2017-05-26:
[Validating Sdf Files](https://bitbucket.org/osrf/sdformat/issues/157/validating-an-sdf-from-the-c-api)

jwnimmer-tri, 2017-06-19:
All issues must have owners.  I'm assigning one arbitrarily.  (Fix it up if I'm wrong.)

PR #6496 (2017-07-20): Double pendulum SDFormat demo
This pull request closes #6135 by parsing a double pendulum model from an SDF file using sdformat into a RigidBodyTree. To this end, some hints were taken from the existing SDF parser (below drake/multibody/parsers/) on how Drake manages bodies' and joints' frames.

@stonier for feature review.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6496)
commit 46b4b829 (2017-07-20): Double pendulum SDFormat demo (#6496)

This pull request closes #6135 by parsing a double pendulum model from an SDF file using sdformat into a RigidBodyTree. To this end, some hints were taken from the existing SDF parser (below drake/multibody/parsers/) on how Drake manages bodies' and joints' frames.
closed 2017-07-20 by jwnimmer-tri