RobotLocomotion/drake #6138: Robust enforcement to use the lower triangle in RotationalInertia

amcastro-tri, 2017-05-18:
As discussed in this [Reviewable comment](https://reviewable.io/reviews/robotlocomotion/drake/5945#-Kjk5x7d8BPlefOZzSIo), the eforcement to use the lower triangular part of the Eigen member in `RotationalInertia` is brittle and there an oportunity for improvement. 

@jwnimmer-tri proposed that we could enforce this all throughout the class in a robust way by placing the Eigen matrix member within a struct with private access. Then that struct would only allow to perform valid operations, like so:
 ```c++
class RotationalInertia {
  public:
    // RotationalInertia methods ...
  private:
   // Internal representation of the rotational inertia as a symmetric matrix.
   // Only access that doesn't break this symmetry is allowed.
   struct SymmetricMatrixRepresentaion {
     /// Access symmetric matrix elements with static assertion.
     template <int i, int j>
     const T& get_value() const {
       static_assert(j <=  i, "Only the lower triangle must be used.");
       return matrix3x3_(i, j);
     }

     /// When indexes cannot be resolved at compile time (as in IsNaN())
     /// you can use this method.
     const T& get_value(int i, int j) const {
       check_and_swap(&i, &j);
       DRAKE_ASSERT(j <=  i);
       return matrix3x3_(i, j);
     }

     // Returns a const Eigen view expression to the symmetric part of the matrix
     // in use by this RotationalInertia.
     const Eigen::SelfAdjointView<const Matrix3<T>, Eigen::Lower>
     get_symmetric_matrix_view() const {
       return matrix3x3_.template selfadjointView<Eigen::Lower>();
     }
     
     // Returns a mutable Eigen view expression to the symmetric part of the
     // matrix in use by RotationalInertia.
     // Note: operator=() is not defined for Eigen::SelfAdjointView and therefore
     // we need to return a TriangularView here.
     Eigen::TriangularView<Matrix3<T>, Eigen::Lower>
     get_mutable_triangular_view() {
       return matrix3x3_.template triangularView<Eigen::Lower>();
      }
    private:
     Matrix3<T> matrix3x3_{Matrix3<T>::Constant(std::numeric_limits<
         typename Eigen::NumTraits<T>::Literal>::quiet_NaN())};
   } I_SP_E_;
};
```

The nice feature of this proposal is that not even developers within the scope of the class could get around this restriction, greatly reducing the chance of introducing errors while nicely serving as documentation at call sites.

sherm1, 2017-05-18:
I like that - definitely an improvement! Does make me wonder if this storage mechanism is worth the trouble though, given Eigen's clunky support for symmetry. An easy alternative would be two Vector3's: (1) the moments (diagonal), and (2) the products (off diagonals). Could slap together a 3x3 Matrix3 when needed.

jwnimmer-tri, 2017-06-19:
All issues must have owners.  I'm assigning one arbitrarily.  (Fix it up if I'm wrong.)

sherm1, 2023-05-23:
This was addressed long ago.

closed 2023-05-23 by sherm1