RobotLocomotion/drake #6139: Use "Make" terminology for static factories consistently in UnitInertia

mitiguy, 2017-05-18:
Per Alejandro's request in PR #5945,
the term "MakeUnitInertia ...." should be used for static factory
methods in UnitInertia class.

sherm1, 2017-05-18:
"Make" seems OK to me but elsewhere we use it for returning heap-allocated objects. Here they are stack allocated. "Construct" would be a reasonable alternative to Make (per Jeremy's suggestion which I adopted in #6038).

amcastro-tri, 2017-05-18:
Actually my suggestion was the opposite, not to use any prefix. So instead of `MakeTriaxiallySymmetric()` just have `TriaxiallySymmetric()`.

jwnimmer-tri, 2017-05-18:
For the record, I also think that "Make" is not only okay, but the best choice for factory methods.

If I've found the correct piece of the 6038 thread, that discussion was about a return-by-value method named "Allocate", which I agree erroneously implies a heap operation.

When using "Make" for factory construction, whether its an automatic or heap object is clear by whether or the method returns a unique_ptr or directly by-value.

sherm1, 2017-05-18:
I'm happy with "Make" also.

jwnimmer-tri, 2024-03-16:
We have a bunch of static factory functions in `UnitInertia` now (`SolidCapsule`, etc.) with the noun form.  We've lived with it for 7 years now.  I don't see the point of deprecating all of those names to add "Make" everywhere.  That seems like busy-work with no real benefit.

closed 2024-03-16 by jwnimmer-tri