RobotLocomotion/drake #6147: Consistent policy for throwing debug-build exceptions for invalid inertias to class UnitInertia

mitiguy, 2017-05-19:
Continue work that has been completed in class RotationalInertia to ensure consistent policy for throwing debug-build exception std::logic_error for invalid inertias.
"Bubble" this up to class UnitInertia (and perhaps SpatialInertia).

Change: In UnitInertia.h, change implementation of 
  UnitInertia<T>& ShiftToCenterOfMassInPlace(const Vector3<T>& p_QBcm_E) 
from   RotationalInertia<T>::MinusEqualsUnchecked(PointMass(p_QBcm_E));
to       RotationalInertia<T>& ShiftToCenterOfMassInPlace(...);

This may also require change to spatial_inertia_tests.cc
GTEST_TEST(SpatialInertia, IsPhysicallyValidWithCOMTooFarOut)
needs to catch std::logic_error.

amcastro-tri, 2017-05-22:
cc'ing @amcastro-tri

jwnimmer-tri, 2017-06-19:
All issues must have owners.  I'm assigning one arbitrarily.  (Fix it up if I'm wrong.)

jwnimmer-tri, 2024-03-16:
Seven years later, it seems like maybe this is not that important -- otherwise we would have done it by now?

Please either close this issue as "not planned", or else add it to a project board for the project that's going to implement it.

closed 2024-03-18 by sherm1