RobotLocomotion/drake #6149: Minimal example of a project using drake as an external library on travis

RussTedrake, 2017-05-20:
bonus points: a second example ising just part of drake (eg just drakeOptimization.lib)

Current pain point is that building drake from source takes >1 hr on travis, so would not be free.  Likely best resolution is to finally provide binary releases.

+ @rdeits, @tkoolen

jamiesnape, 2017-05-22:
Not Jenkins?

stonier, 2017-05-22:
This is for users outside our periphery who are using drake with their own projects and using travis to avoid having to set up their own CI. We just need to provide an appropriate pipeline for distributing versioned binaries for these people.

jamiesnape, 2017-05-22:
Fair enough, though we could do with something like this running on Jenkins at some point.

stonier, 2017-05-22:
Aye, I'm thinking of having a separate repository (repositories) demonstrating in a very simple way a few different use cases. Spinning them up on jenkins would be a good way to test them to make sure it works properly.

jamiesnape, 2017-11-15:
We basically have this in shambhala now using CircleCI, so maybe this can be closed?

jwnimmer-tri, 2018-06-09:
@RussTedrake okay to close now?

closed 2018-06-09 by RussTedrake