RobotLocomotion/drake #6157: Add lane_distance as an additional optional output of api::Lane::ToLanePosition().

jadecastro, 2017-05-22:
## Some Background
Currently, the `distance` argument for `lane->ToLanePosition()` provides a distance to the driveable bounds of a given lane in a particular segment.  The Maliput API states that the driveable bounds are "the lateral bounds for a position that is considered to be "on pavement", reflecting the physical extent of the paved surface of the lane's segment."  

In `monolane` (where there are no adjacent parallel lanes within each segment to reason about) a certain `GeoPosition` is considered to be outside of a given lane if it has `distance > 0` upon calling `monolane::ToLanePosition()`. That is, it is outside of the driveable bounds of the given lane.

In `dragway`, (where it is possible to have multiple adjacent parallel lanes), all lanes share the same pavement in the lateral (`r`) direction, the `driveable_bounds` correspond to the entire width of the dragway itself (inclusive of all the lanes).  Upon calling `dragway::ToLanePosition()`, `distance` is always greater than zero when within any lane of the dragway.  

## Problem
Based on the above, when a user does, for instance, 
```
maliput::api::GeoPosition geo_position{x, y, z};
double distance{};
lane->ToLanePosition(geo_position, nullptr, &distance);
```
it is not necessarily true that a `distance` is positive when `x`, `y`, `z`, are outside of the strict `lane` bounds.  While this is consistent with the API documentation, there are potential use cases where it makes sense to have a similar distance measure to determine if `geo_position` is strictly within `lane`.

For instance, when querying whether or not traffic cars are in the same lane as the ego car (whose `lane` is known), it would be convenient to simply check the condition that `distance > 0`.  The distance might also be useful as a constraint for a lane-keeping controller.

## Proposed Solution
A small addition to the existing API would suffice, that would add `double* lane_distance` as an additional optional argument to `maliput::api::ToLanePosition()`:
```
  /// Determines the LanePosition corresponding to GeoPosition @p geo_position.
  ///
  /// The return value is the LanePosition of the point within the Lane's
  /// driveable-bounds which is closest to @p geo_position (as measured by
  /// the Cartesian metric in the world frame).  If @p nearest_point is
  /// non-null, then it will be populated with the GeoPosition of that
  /// nearest point.  If @p distance is non-null, then it will be populated
  /// with the Cartesian distance from @p geo_position to that nearest point.
  /// If @p lane_distance is non-null, then it will be populated with the Cartesian
  /// distance from @p geo_position to the nearest point that lies within the
  /// @p lane_bounds.
  ///
  /// This method guarantees that its result satisfies the condition that
  /// `ToGeoPosition(result)` is within `linear_tolerance()` of
  /// `*nearest_position`.
  LanePosition ToLanePosition(const GeoPosition& geo_position,
                              GeoPosition* nearest_point,
                              double* distance,
                              double* lane_distance) const {
    return DoToLanePosition(geo_position, nearest_point, distance, lane_distance);
  }
```

liangfok, 2017-05-22:
I'm going to assign this to @maddog-tri to get his blessing. Once received, I'll assign it to myself and will update `dragway` with the requested feature.

maddog-tri, 2017-05-23:
TLDR:  This is what `r` and `lane_bounds()` already do.

Given a lane-frame position `(s,r,h)`, the `r` is the distance off the reference-line for the given lane.  If `r` is positive, then `r - lane_bounds().r_max` tells you how far that position is outside of the lane boundary and, likewise, `lane_bounds().r_min - r` for negative `r`.  If `r` exceeds the lane-bounds, then the "nearest point that lies within the lane bounds" is always `(s, r_max, h)` or `(s, r_min, h)`, respectively.

The API docs for `Lane::driveable_bounds()` do need a little refresh in light of the more nuanced description in the design doc (and that may be one source of confusion here):  

> Note: ...The current definitions of $B_\text{driveable}$ and
$H_\text{driveable}$ conflate the bounds of the /driveable/ volume
(e.g., pavement and free space under bridges) with the bounds of the
/modelled/ volume (e.g., the bounds on $r$ and $h$ which maintain
$G^1$ continuity, avoiding cusps). 

The idea is that `driveable_bounds` is the intersection of "the extent of the asphalt" with "the extent of `r` that allows for a non-screwball `(s,r,h)` parameterization of the lane".  That's why the current `nearest_point` and `distance` exist in `ToLanePosition()`:  for any given non-straight and/or finite length Lane, there are `(x,y,z)` positions in the world-frame that _cannot_ be expressed as an `(s,r,h)` in the lane-frame --- so, `ToLanePosition()` calculates the closest point that _can_ be expressed as an `(s,r,h)` instead (along with the distance to the original point in question).

This statement in "Some Background" above is incorrect:

>In monolane (where there are no adjacent parallel lanes within each segment to reason about) a certain GeoPosition is considered to be outside of a given lane if it has distance > 0 upon calling monolane::Lane::ToLanePosition(). That is, it is outside of the driveable bounds of the given lane.

`monolane` expresses distinct `lane_bounds()` and `driveable_bounds()` --- "outside of a given lane" is any position which exceeds the lane bounds.  Exceeding the driveable bounds does exceed the lane bounds, too --- but one can be "out of lane" without being "off the map".  In other words, `monolane` supports modeling a single-lane road with shoulders.  (`dragway` supports shoulders, too.)

jadecastro, 2017-05-23:
Thanks @maddog-tri.  I realize that I can accomplish the same thing by checking if `r` lies between `lane->lane_bounds().r_min` and `lane->lane_bounds()x.r_max` and also that `s` is positive and less than `lane->length()`.  Ignoring the need to compute any distance to `lane_bounds`, I suppose what API offers suffices and I will make some change to my dev code rather than add anything new to maliput.

(Also- I have removed the monolane comment from the OP.)

closed 2017-05-23 by jadecastro

maddog-tri, 2017-05-23:
W.r.t. "checking if ... and also that `s` is positive and less than `lane->length()`":  the returned `s` value should always be bounded by `[0, length()]`, so you shouldn't have to do that.