RobotLocomotion/drake #6179: TSan issue with MOSEK

m-chaturvedi, 2017-05-25:
A few tests when run with TSan give `FATAL: ThreadSanitizer CHECK failed` error and fail [1].  The error seems to stem from MOSEK.
The error is caused due to the number of mutexes exceeding a certain limit (64) [2] [3].

The tests failing:
- global_inverse_kinematics_collision_avoidance_test
- global_inverse_kinematics_reachable_test
- global_inverse_kinematics_test
- rotation_constraint_test
- rotation_constraint_limit_test

[1] https://drake-cdash.csail.mit.edu/viewDynamicAnalysis.php?buildid=613112
[2] https://github.com/llvm-mirror/compiler-rt/blob/master/lib/sanitizer_common/sanitizer_deadlock_detector.h#L69
[3] https://groups.google.com/forum/#!topic/thread-sanitizer/kbStiqIGVLA

stonier, 2017-05-26:
Thanks for the report @m-chaturvedi

jamiesnape, 2017-11-15:
This is not actually something that we can fix since it is internal to MOSEK. Since we have disabled the tests for TSan builds, I would propose closing this issue.