RobotLocomotion/drake #6197: Some problems with `Value<BasicVector<T>>`

sherm1, 2017-05-29:
As of PR #6187 our type-erasing AbstractValue type `Value<T>` accepts any type T that is copyable or cloneable (previously only copyable worked). This allows `Value<MyVector>` for types MyVector derived from BasicVector, so we no longer have a special hack `VectorValue<T>` for holding those.

I have run into two unexpected consequences of this change while attempting to retrofit output ports to use the new code. I'm not sure either of these are bugs, but wanted to point them out for discussion.

### 1. Definition of is_cloneable
First, `drake::is_cloneable<T>` is very specific about the signature required for T to be cloneable:
```c++
std::unique_ptr<T> Clone() const;
```
It is *not* enough for class T to depend on its base class B for a Clone() method that returns `unique_ptr<B>`. So ... `BasicVector` is considered cloneable, but classes derived from it that implement only `DoClone()` are not. They also need to overload a `Clone()` method something like:
```c++
// class MyVector : public BasicVector<T>
std::unique_ptr<MyVector> Clone() const {
  auto base = BasicVector<T>::Clone();
  return std::unique_ptr<MyVector>(dynamic_cast<MyVector*>(base.release()));
}
```
In particular our automatically-generated BasicVector subtypes like `SimpleCarState` are not cloneable by this definition and hence `Value<SimpleCarState>` doesn't compile. 

It isn't immediately clear to me why `is_cloneable` has this restriction, but it is clearly documented so definitely not an accident. We could attempt to relax that restriction (which would require adding downcasts elsewhere) or just generate concrete-class Clone() methods with the right sig.

### 2. Generic treatment of BasicVectors
Previously `VectorValue` stored concrete BasicVectors using a base class pointer `BasicVector*`, that is the actual AbstractValue containing a `MyVector` was a `Value<BasicVector<T>*>`. So it was possible to take a type-erased AbstractValue known to contain *some* kind of BasicVector and extract the generic BasicVector from it. (Still sort of semi-type erased.) Now however the actual type is `Value<MyVector>`. It is no longer possible to treat that as a generic BasicVector, because `dynamic_cast<Value<BasicVector>>(abstract_value)` fails, since `Value<MyVector>` does not derive from `Value<BasicVector>`.

In general, if class D derives from B it would be useful to obtain a `B&` from an AbstractValue of type `Value<D>`. It's not obvious to me how to do that. Thoughts?

jwnimmer-tri, 2017-05-29:
The two questions are directly related.  The `Value<BasicVector<T>>` must always be used as the type-erased wrapper, even for subclasses of `BasicVector`, so that there is a common type that all vector values share.  It is a feature that `Value<SimpleCarState<T>>` does not compile.

Possibly there is some missing sugar, so that creating typed-erased `SimpleCarState<T>` from scratch is more sensible?

jwnimmer-tri, 2017-06-01:
@sherm1 is there anything else I can do to help on this discussion / issue?

I guess to answer the final question:
> In general, if class D derives from B it would be useful to obtain a B& from an AbstractValue of type Value<D>. It's not obvious to me how to do that. Thoughts?

The same way as prior to the `Value<BasicVector<T>>` -- grab the BasicVector and then downcast:
```c++
class MyVector : public BasicVector<double> { ... };
void foo(const AbstractValue& abstract_value) {
  const MyVector& my_vector = dynamic_cast<const MyVector&>(
      abstract_value.GetValueOrThrow<BasicVector<double>>());
}
```

I have considered making the `AbstractValue::Get...` family of methods have more magic in order to allow for `abstract_value.GetValueOrThrow<MyVector>()` -- by looking up that `MyVector` clones as `BasicVector<double>` and doing two casts instead of one.  Correspondingly, we could offer a factory method to do `MakeValue<MyVector>{args...}`.  I haven't yet tried to do this in code but it should all work.  I was torn about how confusing it would be, and we didn't seem to really need it yet, since mostly the framework is already doing the downcasting in the new LeafSystem sugars.  Most users should not be unpacking AbstractValue manually -- the framework should do it for them and they can rely on their methods being given exactly what they need.

sherm1, 2017-06-02:
Thanks, @jwnimmer-tri. The output port PR uses double-downcasting as you wrote above, and that's fine. But that only works because that code special-cases vector-valued ports so is smart about storing MyVectors as `Value<BasicVectors>`. I don't think Value should have similar special casing -- now that you got rid of VectorValue I am happy that Value is ... um ... value-neutral. 

Where I think we have gone wrong somewhat is the implementation of BasicVector. It would be much nicer if restructured per @soonho-tri's suggestion described [here](https://github.com/RobotLocomotion/drake/issues/3614#issuecomment-279263249). Then BasicVector could be final and have a copy constructor, as could all the other generated vector types. That wouldn't avoid the double downcasting for special handling of vector types, but it would make those types behave more like normal objects.

I'll close this issue as sufficiently discussed.

closed 2017-06-02 by sherm1