RobotLocomotion/drake #6218: Rewrite Monomial class to store Variable instead of Variable::Id

hongkai-dai, 2017-05-31:
Currently a `Monomial` class stores the pair of variable and exponent as
```C++
class Monomial {
 private:
  std::map<Variable::Id, int> powers_;
}
```
The user need to keep the map from `Variable::Id` to `Variable`. We created `Monomial` class in this way with `Variable::Id` as the map key for performance reason. And we thought `Monomial` class is going to be used internally, and the internal user has the responsibility to keep the map from ID to variable.

Now that `Monomial` class is going to be used in Sum-of-Squares optimization by external users, it is easier for the user if `Monomial` class stores `Variable`. So the data member would look like
```C++
class Monomial {
 private:
  std::map<Variable, int, compare_variable> powers_;
}
```
We can then add/rewrite following methods in `Monomial` class
```C++
class Monomial {
 public:
  symbolic::Expression ToExpression() const;
  Monomial operator*(const Monomial& other) const;
  Monomial pow(int exponent) const;
  bool operator==(const Monomial& other) const;
  int total_degree() const;
}
```

hongkai-dai, 2017-05-31:
@soonho-tri

FischerGundlach, 2017-06-01:
Hey, I had a look on `std::map`. Instead of `std::map<Variable, int, compare_variable> powers_;` we could use `std::map<Variable, int, Variable> powers_;`,  and implement an operator  ```bool operator()(const Variable& a, const Variable& b) const { return a.get_id() < b.get_id();  }```.

Is this what you were thinking of for the map, or should `compare_variable` be a seperated class/struct?

soonho-tri, 2017-06-01:
@FischerGundlach, FYI, `symbolic::Variable` already provide `operator<`, it's enough just to have `std::map<Variable, int>` unless we want to customize the ordering.

hongkai-dai, 2017-06-01:
@soonho-tri is right, I just updated the comment on the top.

FischerGundlach, 2017-06-01:
@soonho-tri true. I was skeptical about that name as it implies some monomial is smaller but I will use it.

soonho-tri, 2017-06-06:
Here is my WIP branch: https://github.com/soonho-tri/drake/commits/pr-refactor-monomial. It compiles and all tests passed.

TODO:
 - [ ] write test cases for added functions. (`operator*=` and `pow(int)`)
 - [ ] make `Monomial` compatible with `Eigen::Matrix`

closed 2017-07-18 by soonho-tri