RobotLocomotion/drake #6219: Introduce Polynomial class

hongkai-dai, 2017-05-31:
Currently for Sum-of-Squares optimization, we created the map 
```C++
typedef std::unordered_map<Monomial, Expression, hash_value<Monomial>>
    MonomialToCoefficientMap;
```
to represent polynomial. Since polynomial is going to be used extensively in Sum-of-Squares optimization, we should consider to create a `Polynomial` class that contains the map
```C++
typedef std::unordered_map<Monomial, Expression, hash_value<Monomial>>
    MonomialToCoefficientMap;
class Polynomial {
 private:
  MonomialToCoefficientMap monomial_to_coefficient_map_;
}
```
And the Polynomial class should support operations like `+`, `-`, `*`, `pow`, `decompose`.
```C++
typedef std::unordered_map<Monomial, Expression, hash_value<Monomial>>
    MonomialToCoefficientMap;

class Polynomial {
 public:
  /** Constructs a Polynomial object from a symbolic::Expression object.
  * Decomposes a polynomial `e` into monomials, with respect to a specified set
  * of variables `vars`.
  * A polynomial can be represented as
  * ∑ᵢ c(i) * m(i)
  * where m(i) is a monomial in the specified set of variables, and c(i) is the
  * corresponding coefficient.
  * Note the coefficient will include any constants and symbols not in the set of
  * variables.
  * <pre>
  * Example:
  * For polynomial e1 = 2x²y + 3xy²z + 4z
  * Polynomial(e1, {x,y,z}) will construct a Polynomial object with 
  * monomial_to_coefficient_map[x²y] = 2
  * monomial_to_coefficient_map[xy²z] = 3
  * monomial_to_coefficient_map[z] = 4
  * on the other hand, Polynomial(e1, {x,y}) (notice z is not included in the
  * input argument) will construct a Polynomial object with
  * monomial_to_coefficient_map[x²y] = 2
  * monomial_to_coefficient_map[xy²] = 3z
  * monomial_to_coefficient_map[1] = 4z
  * </pre>
  */
  Polynomial(const Expression& e, const Variables& vars);
  Polynomial(const MonomialToCoefficientMap& monomial_to_coefficient_map);
  Polynomial operator+(const Polynomial& other) const;
  Polynomial operator-(const Polynomial& other) const;
  Polynomial operator*(const Polynomial& other) const;
  Polynomial pow(int exponent) const;

 private:
  MonomialToCoefficientMap monomial_to_coefficient_map_;
}
```
Note that the constructor `Polynomial(const Expression&e, const Variables& vars)` is actually the `DecomposePolynomialIntoMonomials` function we already implemented in https://github.com/RobotLocomotion/drake/blob/master/drake/common/monomial.h#L389
The `+` and `pow` implementation can also be found at https://github.com/RobotLocomotion/drake/blob/master/drake/common/monomial.cc#L207~L254

@soonho-tri @FischerGundlach

soonho-tri, 2017-06-01:
BTW, what's our plan with the existing `Polynomial` class in `drake/common/polynomial.h`? Also /cc @frankpermenter who was interested in the topic.

hongkai-dai, 2017-06-01:
I think they are in different namespace? The `Polynomial` class we want to introduce is in `symbolic` namespace.

jwnimmer-tri, 2017-06-01:
I think Soonho's point would be (or at least mine is) that we would prefer not have _two_ things named Polynomial in the tree, no matter their namespace.

soonho-tri, 2017-06-01:
Yes, I'm afraid that the two classes confuses our users, especially if one is inside of `drake::symbolic` namespace.
 - `drake::Polynomial`
 - `drake::symbolic::Polynomial` (it's the proposed one here).

How about having a discussion next week when we have @hongkai-dai in CAM?

ggould-tri, 2017-12-08:
The confusion created by two Polynomial classes is extracted as its own issue in #7576.

hongkai-dai, 2018-09-25:
Mark as resolved.

closed 2018-09-25 by hongkai-dai