RobotLocomotion/drake #6229: Director in Drake

stonier, 2017-06-01:
Some of the infrastructure plans at the moment are moving towards removing Director as a dependency in the drake bazel build which gets rid of a weird cyclic dependency (some instantiations of director depend on drake and some of drake depend on director).

The only pseudo-dependency that will remain will be a binary download of drake-visualiser to enable visualisation of various demos.

That leaves just a handful of (manipulation) scripts that import director's python:

* `./examples/kuka_iiwa_arm/director_ik_app.py`
* `./multibody/rigid_body_plant/visualization/contact_viz.py`
* `./multibody/rigid_body_plant/visualization/show_time.py`

So, last call for opinions before we act: 

1. retire those scripts (if nobody cares about them) 
2. move them to director
3. move them to a new public repo that would host director based scripts going forward
4. (worst case) don't drop director in drake

Discussed with @sammy-tri on slack and an initial inclination is to opt for 3.

@m-chaturvedi @jamiesnape

SeanCurtis-TRI, 2017-06-01:
Certainly, it's going to be really easy to break these scripts.  For example, `contact_viz.py` depends on lcm message formats defined in drake. Is the plan that we allow the breakage and patch it if/when someone notices? Or do we document the dependency where the changes might occur so that changes in drake will trigger changes in the separate repository?

jwnimmer-tri, 2017-06-01:
There is a distinction between removing Director from the Drake build, and removing Director from Drake CI.  We should make sure that there is a story for CI coverage of relevant combinations of things before rocking the boat too much.

jwnimmer-tri, 2017-06-19:
All issues must have owners.  I'm assigning one arbitrarily.  (Fix it up if I'm wrong.)

jwnimmer-tri, 2018-06-09:
Moving owners to match #7024.

EricCousineau-TRI, 2018-11-01:
Marking this as resolved by #9653 and that we have not put dropping director onto our current roadmap.
Can reopen if there's another resolution criteria

closed 2018-11-01 by EricCousineau-TRI