RobotLocomotion/drake #6237: build make issue  ubuntu 16.04

hughhugh, 2017-06-02:
Hi, I am trying to complie the drake in ubuntu 16.04, matlab 2017a , but I got some errors , I have tried two methods except ROS. Coud any one figure it out ? The matlab works by using command `sudo matlab`. Thank you.
logs as the following:
[ 81%] Performing configure step for 'sedumi'
-- The C compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMake Error at /usr/share/cmake-3.5/Modules/FindMatlab.cmake:1029 (list):
  list GET given empty list
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindMatlab.cmake:1235 (_Matlab_get_version_from_root)
  CMakeLists.txt:5 (find_package)
-- Found Matlab: /usr/local/MATLAB/R2017a/extern/include  found components:  MAIN_PROGRAM 
-- Configuring incomplete, errors occurred!
See also "/home/name/drake_distro/build/externals/sedumi/CMakeFiles/CMakeOutput.log".
CMakeFiles/sedumi.dir/build.make:106: recipe for target 'sedumi-prefix/src/sedumi-stamp/sedumi-configure' failed
make[2]: *** [sedumi-prefix/src/sedumi-stamp/sedumi-configure] Error 1
CMakeFiles/Makefile2:6352: recipe for target 'CMakeFiles/sedumi.dir/all' failed
make[1]: *** [CMakeFiles/sedumi.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2

jwnimmer-tri, 2017-06-19:
All issues must have owners.  I'm assigning one arbitrarily.  (Fix it up if I'm wrong.)

RussTedrake, 2017-08-01:
can you please try again (with master) and reopen if it's still a problem?  sedumi -- the particular package that gave you this error -- is no longer used in drake.

closed 2017-08-01 by RussTedrake