RobotLocomotion/drake #6240: Refactor CI 'experimental' job names

stonier, 2017-06-02:
These jobs confusingly consist of both pre-merge and defacto experimental jobs. For pre-merge jobs do a substitution "experimental"->"premerge".

Update the CI design doc [here](https://docs.google.com/document/d/1oPybZVy5KlPb9IY08t6ol_l-oakXRMV9rmWmpsI_8aw/edit#bookmark=id.i96xhsqarv2) once done.

jamiesnape, 2017-06-02:
This is extremely non-trivial and will also break the GitHub statuses of open PRs.

jamiesnape, 2017-06-02:
It will also break links back to Jenkins from CDash on existing builds.

stonier, 2017-06-02:
Was this visited in the past and discarded with David at some point because of the complications?

jamiesnape, 2017-06-02:
It was discussed, yes. There are probably ways we can make the PR jobs more prominent, but the names themselves are pretty much immutable.

EricCousineau-TRI, 2018-01-18:
Will close this for now, as it would cause disruption. Can reopen later if needed.

closed 2018-01-18 by EricCousineau-TRI