RobotLocomotion/drake #6247: nice_type_name.cc Defines Regular Expressions that are Incompatible with gcc 4.8.5

liangfok, 2017-06-04:
`drake::NiceTypeName::Canonicalize(const string& demangled)` currently defines several regular expressions that are incompatible with the `std::regex` provided by gcc 4.8.5. This results in a `std::regex_error` being thrown.

[Here is the current code.](https://github.com/RobotLocomotion/drake/blob/ee75498908a078a4d641757af870ce351bd51a71/drake/common/nice_type_name.cc#L52-L69)

We should update the code to gracefully handle this situation.

rdeits, 2017-06-05:
Possibly relevant: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=53631 

Regexes in GCC 4.8 basically just don't work at all. The code will generally compile, but will silently do the wrong thing (for example, all regex::match calls just return `false`). It's not so much a library as a trap.

jwnimmer-tri, 2017-06-05:
Hah, that is a great characterization!  In its most recent form, PR #6248 resolves this issue by yielding a more specific runtime_error when someone tries to use NiceTypeName with GCC 4.8.  Since Drake only supports GCC >= 4.9 officially, this is probably good enough.

PR #6248 (2017-06-05): Modifies drake::NiceTypeName::Canonicalize() to throw exception when compiled with gcc 4.8 or below.
Resolves #6247.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6248)
closed 2017-06-05 by liangfok