RobotLocomotion/drake #6254: Collision detection return more than a single point for contacting geometries

avalenzu, 2017-06-05:
Currently `ComputeMaximumDepthCollisionPoints` returns a single `PointPair` for each pair of geometries. This makes it impossible to simulate face-face contact and leads to workarounds like zero-radius spheres on the corners of boxes.

Ideally, collision detection should return the contact patch between two bodies. A less ambitious possibility that would resolve most of the cases above would be to return multiple contact points for a pair with face-face contact.

sherm1, 2017-06-05:
:+1: Note that when we use a compliant contact response model we need continuity in the contact point set. In that case each point is representing a contact area and the resulting force at that point will be an indepenent function of that area and the penetration depth. Changes in number or placement of points between steps can cause force discontinuities. Ideally, new points would appear and evaporate only at zero penetration, and the geometry engine would provide appropriate area weightings (or a patch representation). 

That isn't a problem for rigid contact because the forces at all points are solved simultaneously, with the total force distributed among whatever points happen to be there.

SeanCurtis-TRI, 2017-06-05:
This is definitely something that has been bedeviling us for a while. The `ContactResults` output port (on `RigidBodyTree`) was architected such that it assumed the contact manifold would ultimately be more than just a single point. Right now, it's definitely being underserved by the results we're producing.

Of course, then you get into the interesting question of what does this "patch" look like? Geometrically, we have a volume with a discontinuous boundary that is piecewise-linear for mesh-mesh contact but gets far more interesting when we intersect arbitrary geometric shapes. Once we get the infrastructure in, I'm looking forward to tackling this kind of thing.

EricCousineau-TRI, 2017-06-10:
Posting for reference: see **Contact Samples** in [**Working with Contacts in Drake**](http://drake.mit.edu/doxygen_cxx/group__contact__engineering.html)

jwnimmer-tri, 2019-10-10:
@sherm1 or @avalenzu Is this still relevant as a tracking issue?  Perhaps this could be resolved as a duplicate of some more recent work?

avalenzu-tri, 2019-10-10:
I'll let @sherm1 weigh in on whether they have another, more recent tracking issue. No objection to closing this from my end.

sherm1, 2019-10-10:
I am happy to say we are close to having a resolution of this problem with hydroelastic contact in Drake. There are two "Epic" issues tracking that: #9905 and #11170 so we can close this issue.

With hydroelastics, each contact is represented by a triangulated "contact surface". There is a minimum of one triangle, but each triangle is sampled at three quadrature points. So there are always at least 3 "contact points" where forces are generated, with the result weighted by the triangle area.

N.B. Rigid/rigid contacts will continue to be modeled using the current method which is limited to one contact point per geometry pair. But if at least one of the geometries is compliant then rich contact results will be available.

closed 2019-10-10 by sherm1