RobotLocomotion/drake #6295: RgbdCamera's texture rendering is not working correctly with VTK 7.1

kunimatsu-tri, 2017-06-08:
`RgbdCamera`'s texture rendering is not correctly working with VTK 7.1 which is introduced by #6152.
I think the fact that [we had to disable an unit test](https://github.com/kunimatsu-tri/drake/blob/master/drake/systems/sensors/test/rgbd_camera_test.cc#L466) is related to that.

Drake's `RgbdCamera` is simply using `vtkPNGReader` and `vtkOBJReader` to render mesh with texture.

I wrote [a simple rendering program](https://github.com/kunimatsu-tri/sandbox/tree/master/vtk/RenderTexture) apart from Drake, and found that the same problem is still happening. So, the cause is not from Drake, but from VTK.  I recorded a video for that: https://youtu.be/tYGpk80CyZI  You can see "hole" like visualization on the bottle mesh in this video.

Also, I confirmed that this happens with VTK 7.0 too, but did NOT happen with VTK 5.8 nor VTK 6.1.

SeanCurtis-TRI, 2017-06-08:
Your normals are reversed. The big difference might simply be that your normals have *always* been reversed but the new version of vtk has finally enabled back-face culling. Can you inject arbitrary opengl commands into the pipeline?

kunimatsu-tri, 2017-06-08:
Thanks @SeanCurtis-TRI. I first doubted it, but I couldn't find any normals that are reversed besed-on meshlab's visualization:
[image]

SeanCurtis-TRI, 2017-06-08:
That image *seems* to indicate the normals aren't reversed. However, radial symmetry can be misleading (because I've seen normal visualizations where the normals are 'unit" length, but the object is much smaller than unit length, and so the normals pass all the way through the geometry and stick out the opposite side, closely resembling what you would expect to see.)  However, the normals at the bottom are promising. A definitive visual test would be to visualize it wireframe mode to confirm that no normals are intersecting the center.

There's another (less likely) possibility. For some odd reason, it may simply be rendering the back faces instead of the front faces.

Where is this model so I can look at it myself, see if there's anything about the model itself that is giving that behavior.

kunimatsu-tri, 2017-06-08:
I also confirmed that there is no normal pointing inwards:
[image]

This is in google-drive. I'll send the pointer to you.

Thanks!

SeanCurtis-TRI, 2017-06-08:
I went back and watched the video. While it *generally* seems to exhibit the trait of reversed normals, there are times where it seems they aren't. I believe this is the "hole-like visualization" to which you alluded. So, I have some more ideas:

1. Does the texture image have any alpha channel? (I suspect not.)
2. What is your view frustum set to? Has the accuracy of your depth buffer changed?

kunimatsu-tri, 2017-06-08:
> 1. Does the texture image have any alpha channel? (I suspect not.)

Yes, it does. Here is the output from pnginfo:
```
  Image Width: 1024 Image Length: 1024
  Bitdepth (Bits/Sample): 8
  Channels (Samples/Pixel): 4
  Pixel depth (Pixel Depth): 32
  Colour Type (Photometric Interpretation): RGB with alpha channel 
  Image filter: Single row per byte filter 
  Interlacing: No interlacing 
  Compression Scheme: Deflate method 8, 32k window
  Resolution: 2835, 2835 (pixels per meter)
  FillOrder: msb-to-lsb
  Byte Order: Network (Big Endian)
  Number of text strings: 0 of 0
```

>  2. What is your view frustum set to? Has the accuracy of your depth buffer changed?

This happens at the same camera pose and view frustum as VTK 5.8's case, but only happens with VTK 7.1 (and 7.0). I don't touch any depth buffer setting for this.

I sent the pointer to the model by email.

SeanCurtis-TRI, 2017-06-08:
For the sake of completeness -- what are the view frustum values?

sankhesh, 2017-06-08:
Is this using the OpenGL or OpenGL2 backend?

SeanCurtis-TRI, 2017-06-08:
A couple of quick things you can try:

1. Edit the mtl file and remove the `map_Kd` line.
2. Edit the mtl file and replace `disposable_bottle_pet_tea.png`, with a alpha-less color gradient.
3. Does the phenomenon change at all closer/further from the view point?

kunimatsu-tri, 2017-06-08:
> what are the view frustum values?

In the `RgbdCamera`, view angle for y (vertical direction) is 45 degrees, and the clipping range is 0.001 to 100.0. I don't set any those values for the example code though (the video is from my example code).

> Is this using the OpenGL or OpenGL2 backend?

I think OpenGL2 backend since this is what we can find at build configuration files:
Drake bazel:
https://github.com/RobotLocomotion/drake/blob/master/drake/systems/sensors/BUILD#L183

My example code:
https://github.com/kunimatsu-tri/sandbox/blob/master/vtk/RenderTexture/CMakeLists.txt#L11

> A couple of quick things you can try:

Actually, I'm not using mtl file at all at my example code nor `RgbdCamera`.

Can you run my code at your environment?
https://github.com/kunimatsu-tri/sandbox/tree/master/vtk/RenderTexture

SeanCurtis-TRI, 2017-06-08:
First the sad answer -- I'm assuming that it's going to take me a fair amount of effort to run your code on my machine -- I've never used VTK and so I'd have to get everything configured.

1. When you say you "don't set any those values for the example code", does that mean you don't know what the values for the frustum are in the example code? Or that the example code uses those defaults? The frustum is 100m deep, and the bottle is 7-8 cm deep. Still, that's only three orders of magnitude and its in the front of the frustum. So, it's probably not a depth buffer issue.
2. If you're not using the mtl file, how is the object and texture map associated with each other?  Explicitly in code? In any event, the goal is to remove the texture to see if it's texture rendering that's contributing to the problem. If you have the same problem with only materials, then we can factor out texture-based causes.

kunimatsu-tri, 2017-06-08:
Ah OK, that's fine.

No.1:
It should be using default values. I didn't care in my example code simply because it's interactive, so you can change the depth and pose directory by mouse operation. I confirmed that with VTK 5.8, the "hole" like visualization never happened at various poses and depths while it did with VTK 7.1.

No.2:
I was assuming it is handled by the VTK's classes, namely vtkTexture and vtkActor. That's why I'm asking this to Kitware people.

I have to leave now but happy to answer to any questions later this evening.

sankhesh, 2017-06-09:
@kunimatsu-tri I think it is a depth peeling issue. You should be able to reproduce it even without a texture map, right?

kunimatsu-tri, 2017-06-09:
@sankhesh Thanks for your working on this issue.

I could NOT reproduce this without a texture (I tried this at various view points and poses):
[image]

I modified my example code so that you can easily try turning on/off texture:
https://github.com/kunimatsu-tri/sandbox/tree/master/vtk/RenderTexture

sankhesh, 2017-06-09:
That's odd. Let me see if I find something.

SeanCurtis-TRI, 2017-06-09:
Try creating another texture that is just a solid image (make sure it has no alpha channel). What does that do?

sankhesh, 2017-06-09:
Great point @SeanCurtis-TRI. Even without a texture, could you try decreasing the opacity by `actor->GetProperty()->SetOpacity(0.8);`

kunimatsu-tri, 2017-06-09:
`actor->GetProperty()->SetOpacity(1.0);` still happening.
[image]

`actor->GetProperty()->SetOpacity(0.8);` still happening.
[image]

sankhesh, 2017-06-09:
Sorry @kunimatsu-tri I meant remove texture map and reduce opacity.

kunimatsu-tri, 2017-06-09:
@sankhesh I sent the model to you by email, can you try? since I'm not sure what that exactly means.

sankhesh, 2017-06-09:
@kunimatsu-tri Yep, you didn't have depth peeling enabled.

[image]

Here is how I enabled depth peeling in your mock code:

```cpp
diff --git a/vtk/RenderTexture/RenderTexture.cxx b/vtk/RenderTexture/RenderTexture.cxx
index 7596d09..f19829e 100644
--- a/vtk/RenderTexture/RenderTexture.cxx
+++ b/vtk/RenderTexture/RenderTexture.cxx
@@ -60,11 +60,14 @@ int main(int argc, char* argv[]) {
 
   vtkNew<vtkRenderer> renderer;
   renderer->AddActor(actor.GetPointer());
+  renderer->SetUseDepthPeeling(1);
   renderer->SetBackground(0.9, 0.5, 0.1);
 
   vtkNew<vtkRenderWindow> renderWindow;
   renderWindow->AddRenderer(renderer.GetPointer());
   renderWindow->SetSize(640,480);
+  renderWindow->SetAlphaBitPlanes(1);
+  renderWindow->SetMultiSamples(0);
 
   vtkNew<vtkRenderWindowInteractor> renderWindowInteractor;
   renderWindowInteractor->SetRenderWindow(renderWindow.GetPointer());
```

kunimatsu-tri, 2017-06-09:
@sankhesh Wonderful! Thanks a lot!
I'll submit a PR for `RgbdCamera` with this technique.

In the meantime, I wonder If there would be any document which explains why the default behavior changed since VTK 7.0. Is it because of the backend change (OpenGL -> OpenGL2)?

sankhesh, 2017-06-09:
> In the meantime, I wonder If there would be any document which explains why the default behavior changed since VTK 7.0. Is it because of the backend change (OpenGL -> OpenGL2)?

I am not sure why it would have worked with VTK 6 since depth peeling has always been disabled by default (as far as I know). Don't think the backend change had anything to do with it.

SeanCurtis-TRI, 2017-06-09:
I've learned something new. I wasn't aware of depth peeling.  :)

kunimatsu-tri, 2017-06-09:
Yup, I've learned it here too!

However, when I tried depth peeling with `RgbdCamera`, I found the contour of rendered object became jaggy.  I think this is a side effect from `renderWindow->SetMultiSamples(0);`

I wonder if we could render texture that has alpha channel correctly without having any side effects like this.

Depth peeling enabled:
[image]

Depth peeling disabled (default):
[image]

SeanCurtis-TRI, 2017-06-09:
Are you *really* planning on having transparent objects? Given that vision struggles with that, I don't know that we want to also throw in the rendering challenges as well. Seems to me, your default case should be no transparency.

sankhesh, 2017-06-09:
> However, when I tried depth peeling with RgbdCamera, I found the contour of rendered object became jaggy. I think this is a side effect from renderWindow->SetMultiSamples(0);

Yep.

> I wonder if we could render texture that has alpha channel correctly without having any side effects like this

`renderwindow->SetMultiSamples()` uses MSAA. Combining that with depth peeling is not a good idea. 
You could try the new FXAA approach with depth peeling. See the [FXAA in VTK](https://blog.kitware.com/new-fxaa-anti-aliasing-option-in-paraviewvtk/) blog post for more details.
Essentially, you'd use `vtkRenderer::UseFXAAOn()`

kunimatsu-tri, 2017-06-09:
> Are you really planning on having transparent objects?

That's a good point. Assuming no transparency is definitely one thing we can do for now.
Sooner or later, we will need it though since we can't ignore the existence of transparent objects.

Also, I was still wondering why this weird behavior happens despite the fact that the texture has 255 for the alpha value which means no transparency.

It would be nice if we could find a way to handle such a case correctly.  If this would be blocking anything urgent, please let me know.

> You could try the new FXAA approach with depth peeling. See the FXAA in VTK blog post for more details.
Essentially, you'd use vtkRenderer::UseFXAAOn()

Oh, thanks! I will. What a fruitful day today is.

kunimatsu-tri, 2017-06-09:
With `UseFXAAOn()`, I've got this image which is good!

[image]

I'm gonna move on to a PR with this. Thanks both @sankhesh and @SeanCurtis-TRI for taking your time!

closed 2017-06-12 by kunimatsu-tri