RobotLocomotion/drake #6296: Supported macOS / OS X versions for binaries

jamiesnape, 2017-06-08:
We currently support Yosemite, El Capitan, and Sierra for binaries, and will inevitably need to support High Sierra in a couple of months as people upgrade or buy new Macs. There is a maintenance cost to all this (we have to maintain 3-4 different VM's on which to create binaries).

The question therefore is which versions should we support?

Note that soon Director and VTK will only be directly available from the Bazel build in binary form. We do not support source builds of those dependencies yet. We will also distribute binary releases of Drake in the next few weeks, so the question applies to those binaries too.

soonho-tri, 2017-06-08:
Here is a survey: https://www.gosquared.com/global/mac/sierra/#month

It seems that supporting the latest three versions (Sierra, El Capitan, and Yosemite) cover about 90% of users. FYI, [bazel.rb](https://github.com/Homebrew/homebrew-core/blob/master/Formula/bazel.rb) only provides pre-built binaries for those three versions.

BTW, how about moving to `Homebrew/homebrew-core` so that we can use their infrastructure to prepare binaries?

stonier, 2017-06-08:
@soonho-tri what are you and Russ using? Are there other mac users in our group itself?

jamiesnape, 2017-06-08:
> BTW, how about moving to Homebrew/homebrew-core so that we can use their infrastructure to prepare binaries?

We probably could for VTK eventually, for the others, it unlikely.

soonho-tri, 2017-06-08:
@stonier , I'm using the latest (Sierra, 10.12.5). I usually do the upgrade after a month of a release.

jamiesnape, 2017-06-08:
FWIW, I use Sierra too and upgrade a week or so after release on my work machine, earlier at home.

jamiesnape, 2017-06-08:
Not enough users and it is mostly a GUI.

stonier, 2017-06-09:
I'd like to propose an N, N-1 support policy, with a small buffer for transition as a new release comes out. That should satisfy the majority of developers. Does anyone object?

soonho-tri, 2017-06-09:
I like the `N` & `N-1` policy.

Speaking of the macOS versions, do we plan to upgrade the macCI from El Capitan to Sierra? I vaguely remember that there are some issues.

jamiesnape, 2017-06-09:
Sounds good to me. I would propose deprecating the N-2 one month after a new N release, removing support after three months. CI would move to the N release within those three months. That would probably mean that El Capitan support would be dropped by the end of this calendar year.

Unless someone comments that they need Yosemite, I would like to drop support ASAP.

jamiesnape, 2017-06-09:
(@soonho-tri Yes, soon, VMware have caught up with support now.)

rdeits, 2017-06-09:
I use El Capitan but am willing to upgrade if necessary (I'm not that interested in features, so I tend to upgrade roughly whenever a program I care about requires it)

RussTedrake, 2017-06-12:
i'm on sierra and support N,N-1

PR #7456 (2017-11-10): Update supported Mac platforms
El Capitan is gone, so OS X is also gone. Closes #6296.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7456)
closed 2017-11-10 by jamiesnape