RobotLocomotion/drake #6317: error when build on ubuntu 14.04

mzwang828, 2017-06-10:
Hey,
I'm trying to install Drake from source using make on Ubuntu 14.04 with matlab 2016 but met this error ‘const class octomap::OcTree’ has no member named ‘nodeHasChildren’

`[ 40%] Building CXX object src/CMakeFiles/fcl.dir/narrowphase/continuous_collision.cpp.o
In file included from /opt/ros/indigo/include/octomap/OcTreeBaseImpl.h:45:0,
                 from /opt/ros/indigo/include/octomap/OccupancyOcTreeBase.h:44,
                 from /opt/ros/indigo/include/octomap/OcTree.h:38,
                 from /opt/ros/indigo/include/octomap/octomap.h:37,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/geometry/octree/octree.h:48,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/detail/traversal/octree/collision/mesh_octree_collision_traversal_node.h:46,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/detail/collision_func_matrix-inl.h:73,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/detail/collision_func_matrix.h:86,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/collision-inl.h:43,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/collision.h:68,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/continuous_collision-inl.h:47,
                 from /home/mzwang/drake-distro/externals/fcl/src/narrowphase/continuous_collision.cpp:38:
/opt/ros/indigo/include/octomap/OcTreeKey.h:52:4: warning: extra ‘;’ [-Wpedantic]
   };
    ^
In file included from /opt/ros/indigo/include/octomap/OccupancyOcTreeBase.h:45:0,
                 from /opt/ros/indigo/include/octomap/OcTree.h:38,
                 from /opt/ros/indigo/include/octomap/octomap.h:37,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/geometry/octree/octree.h:48,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/detail/traversal/octree/collision/mesh_octree_collision_traversal_node.h:46,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/detail/collision_func_matrix-inl.h:73,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/detail/collision_func_matrix.h:86,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/collision-inl.h:43,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/collision.h:68,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/continuous_collision-inl.h:47,
                 from /home/mzwang/drake-distro/externals/fcl/src/narrowphase/continuous_collision.cpp:38:
/opt/ros/indigo/include/octomap/AbstractOccupancyOcTree.h:238:2: warning: extra ‘;’ [-Wpedantic]
 }; // end namespace
  ^
In file included from /home/mzwang/drake-distro/externals/fcl/include/fcl/geometry/octree/octree.h:145:0,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/detail/traversal/octree/collision/mesh_octree_collision_traversal_node.h:46,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/detail/collision_func_matrix-inl.h:73,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/detail/collision_func_matrix.h:86,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/collision-inl.h:43,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/collision.h:68,
                 from /home/mzwang/drake-distro/externals/fcl/include/fcl/narrowphase/continuous_collision-inl.h:47,
                 from /home/mzwang/drake-distro/externals/fcl/src/narrowphase/continuous_collision.cpp:38:
/home/mzwang/drake-distro/externals/fcl/include/fcl/geometry/octree/octree-inl.h: In member function ‘fcl::OcTree<S>::OcTreeNode* fcl::OcTree<S>::getNodeChild(fcl::OcTree<S>::OcTreeNode*, unsigned int)’:
/home/mzwang/drake-distro/externals/fcl/include/fcl/geometry/octree/octree-inl.h:179:16: error: ‘const class octomap::OcTree’ has no member named ‘getNodeChild’
   return tree->getNodeChild(node, childIdx);
                ^
/home/mzwang/drake-distro/externals/fcl/include/fcl/geometry/octree/octree-inl.h: In member function ‘const OcTreeNode* fcl::OcTree<S>::getNodeChild(const OcTreeNode*, unsigned int) const’:
/home/mzwang/drake-distro/externals/fcl/include/fcl/geometry/octree/octree-inl.h:191:16: error: ‘const class octomap::OcTree’ has no member named ‘getNodeChild’
   return tree->getNodeChild(node, childIdx);
                ^
/home/mzwang/drake-distro/externals/fcl/include/fcl/geometry/octree/octree-inl.h: In member function ‘bool fcl::OcTree<S>::nodeChildExists(const OcTreeNode*, unsigned int) const’:
/home/mzwang/drake-distro/externals/fcl/include/fcl/geometry/octree/octree-inl.h:203:16: error: ‘const class octomap::OcTree’ has no member named ‘nodeChildExists’
   return tree->nodeChildExists(node, childIdx);
                ^
/home/mzwang/drake-distro/externals/fcl/include/fcl/geometry/octree/octree-inl.h: In member function ‘bool fcl::OcTree<S>::nodeHasChildren(const OcTreeNode*) const’:
/home/mzwang/drake-distro/externals/fcl/include/fcl/geometry/octree/octree-inl.h:214:16: error: ‘const class octomap::OcTree’ has no member named ‘nodeHasChildren’
   return tree->nodeHasChildren(node);
                ^
make[5]: *** [src/CMakeFiles/fcl.dir/narrowphase/continuous_collision.cpp.o] Error 1
make[4]: *** [src/CMakeFiles/fcl.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [fcl-prefix/src/fcl-stamp/fcl-build] Error 2
make[1]: *** [CMakeFiles/fcl.dir/all] Error 2
make: *** [all] Error 2
`

Could you give me a clue on this? Thanks!

jamiesnape, 2017-06-12:
Try `sudo apt-get remove ros-indigo-octomap ros-indigo-fcl` and then rebuild.

mzwang828, 2017-06-13:
That solves the problem. Thanks!

closed 2017-06-15 by EricCousineau-TRI