RobotLocomotion/drake #6321: ignition_math version configuration hard-coded and not autmoatically kept in sync

fbudin69500, 2017-06-12:
To configure config.hh, ignition_math version number is hard-coded. Therefore when updating ignition_math version in WORKSPACE, one has to think to also update the version number in tools/ignition_math.BUILD .
This is due to a limitation of cmake_configure_file() which does support CMake string operations to compute CMake variable values.

Better CMake parsing could be implemented based on examples [[1]](https://github.com/mwoehlke-kitware/Slicer/blob/master/Utilities/Scripts/SlicerWizard/CMakeParser.py) and [2](https://github.com/mwoehlke-kitware/Slicer/blob/master/Utilities/Scripts/SlicerWizard/ExtensionProject.py) to handle more complex CMake parsing and avoid mismatch library version number due to version update.

mwoehlke-kitware, 2017-06-12:
FYI, if/when we do this, we should investigate all uses of `cmake_configure_file`; there are at least a couple that would probably benefit.

jwnimmer-tri, 2018-06-09:
Since the hard-coded numbers lack comment reminders, I think I screwed this up in my monthly bump for externals version in the path few months.  I'll take this on to repair or at least mitigate for now next time I do the monthly update, and then close out this issue.

jwnimmer-tri, 2018-06-14:
In #8975, I've added some reminder comments, which is good enough for me.  Bumping back over to kitware again now.

jwnimmer-tri, 2018-11-14:
Doesn't need a tracking issue.  If it gets painful beyond our existing mitigations, we'll fix it at that moment.

closed 2018-11-14 by jwnimmer-tri