RobotLocomotion/drake #6341: Support QuadraticConstraint in MathematicalProgram

hongkai-dai, 2017-06-15:
Gurobi natively supports the convex quadratic constraint `x'*Q*x + q' * x <= b`, we should add that into `MathematicalProgram` and `GurobiSolver`.

RussTedrake, 2021-06-15:
Looks like Mosek supports them, too.  It seems that we have [QuadraticConstraint](https://drake.mit.edu/doxygen_cxx/classdrake_1_1solvers_1_1_quadratic_constraint.html) and the solvers declare if they support quadratic constraints, but we just don't have `AddQuadraticConstraint` nor `AddConstraint<Binding<QuadraticConstraint>>` on `MathematicalProgram`?

hongkai-dai, 2021-06-15:
Yes, and that is intentional. Mosek strongly suggests the users to reformulate the quadratic constraints as Lorentz cone constraints, mentioned [here](https://docs.mosek.com/9.2/rmosek/prob-def-quadratic.html#a-recommendation).

hongkai-dai, 2021-06-15:
If it helps, I think we could provide this function
```cc
AddQuadraticConstraint(const Expression& quadratic_expr, double upper_bound, bool lorentz_cone=true);
```
which impose the constraint `quadratic_expr <= upper_bound`. And by default we enforce this constraint as a Lorentz cone constraint (with `lorentz_cone=true`).

RussTedrake, 2021-06-15:
That could be nice, but it's certainly not urgent.

Should we perhaps delete [QuadraticConstraint](https://github.com/RobotLocomotion/drake/blob/master/solvers/constraint.h#L188) and [kQuadraticConstraint](https://github.com/RobotLocomotion/drake/blob/8fed7d2f716c5bb5f3eac1999b4786b484b26760/solvers/program_attribute.h#L18) if they are unused and if we have no intention of using them?

hongkai-dai, 2021-06-15:
Currently QuadraticConstraint and kQuadraticConstraint are used in nonlinear optimization, for example by SnoptSolver. So I think it is useful to keep it around?

RussTedrake, 2021-06-16:
How does a user create one of those constraints?

hongkai-dai, 2021-06-16:
The user can do
```
prog.AddConstraint(std::make_shared<QuadraticConstraint>(...), vars)
```
as shown in snopt_solver_test.cc https://github.com/RobotLocomotion/drake/blob/6c5a106b1d3660a5e10c3667af4eddaf218a479c/solvers/test/snopt_solver_test.cc#L99-L102

Just curious, do you have a convex optimization problem with a convex quadratic constraint? I think that's a common use case, and I can add the function to parse the convex constraint to a lorentz cone constraint.

RussTedrake, 2021-06-16:
I have one in my open PR: https://github.com/RobotLocomotion/drake/pull/15194/files#diff-eeb9eea73750e4da6b48cbda49815973217d01264b9ed577ea725ba477b2720bR95

hongkai-dai, 2023-03-05:
As mentioned in https://github.com/RobotLocomotion/drake/issues/18940#issuecomment-1455154129.

I think I should add an API

```
MathematicalProgram::AddQuadraticConstraint(...);
```
which adds `Binding<QuadraticConstraint>` into MathematicalProgram. Then the solver will either treat it like a generic constraint (just compute the value and the gradient), or a Lorentz cone constraint (by convex solvers like Mosek, Gurobi, SCS, etc).

This will involve quite some change in MathematicalProgram, as now we will add a new data member
```
std::vector<Binding<QuadraticConstraint>> quadratic_constraints_;
```
into MathematicalProgram. Also we need to mark each QuadraticConstraint whether it is convex or not.

hongkai-dai, 2023-06-21:
Prioritizing this because we are solving QCQP.