RobotLocomotion/drake #6350: IpoptSolver counts "acceptable" solutions as SolutionResult::kUnknownError

jefesaurus, 2017-06-15:
See https://www.coin-or.org/Ipopt/documentation/node36.html
And https://github.com/coin-or/Ipopt/blob/8d0b039db68fce54b9bb9bc1d143fa8db88308fc/Ipopt/src/Interfaces/IpAlgTypes.hpp

By setting acceptability options the return result can be `Solved_To_Acceptable_Level` the exact return type I think is `Ipopt::STOP_AT_ACCEPTABLE_POINT` which I believe should be counted as `SolutionResult::kSolutionFound` or something other than and Error result

jwnimmer-tri, 2017-06-19:
All issues must have owners.  I'm assigning one arbitrarily.  (Fix it up if I'm wrong.)

sammy-tri, 2017-06-19:
I added myself as an assignee but I don't expect to get to this soon.  Will post here if I start working on it.

jefesaurus, 2017-06-22:
#6413

sammy-tri, 2019-02-14:
Fixed in #10345

closed 2019-02-14 by sammy-tri