RobotLocomotion/drake #6355: LCM and ROS systems should appear in doxygen groups

RussTedrake, 2017-06-16:
http://drake.mit.edu/doxygen_cxx/group__systems.html

I would like to see a subgroup for "Message Passing" that contains the leaf systems in 
https://github.com/RobotLocomotion/drake/tree/159b31408998aea34299913843d12f49e81e3d45/drake/systems/lcm

and (if it makes sense)
https://github.com/RobotLocomotion/drake/tree/159b31408998aea34299913843d12f49e81e3d45/ros/drake_ros_systems/src

liangfok, 2017-06-16:
I can add the LCM systems to a "Message passing" subgroup but cannot add the ROS systems because Doxygen only scans the source code in `drake-distro/drake/`:

https://github.com/RobotLocomotion/drake/blob/f71c8a52dcdb48664a7828632789c8633bbdb262/drake/doc/doxygen.py#L34

@jwnimmer-tri, can `doxygen.py` be modified to also scan `drake-distro/ros/`?

jwnimmer-tri, 2017-06-16:
The configuration file is at https://github.com/RobotLocomotion/drake/blob/master/drake/doc/Doxyfile_CXX.in.

liangfok, 2017-06-16:
Thanks, I modified the configuration file to include `drake-distro/ros` in #6356.

PR #6356 (2017-06-16): Adds a "Message Passing" Doxygen group containing LCM systems.
Resolves #6355. Here is what the generated Doxygen website looks like:

[image]

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6356)
closed 2017-06-16 by sherm1