RobotLocomotion/drake #6396: IsolateWitnessTriggers fails due to mistaken attempt to repeat integration step

sherm1, 2017-06-21:
### Symptom
PR #6265 attempted to introduce Jacobian re-use in the implicit integrator, a conventional technique that provides substantial performance improvement. That caused `Simulator::IsolateWitnessTriggers()` to fail: a witness function that had previously been seen to cross zero during event isolation failed to cross zero when recomputed, triggering a fatal bug catcher.

### Cause
Implicit integration requires convergence of a nonlinear solver as an intermediate computation. User-specified accuracy controls to what tolerance the nonlinear system must be solved; the exact result is different depending on numerical details including the in-use Jacobian (though always below tolerance). `Simulator::IsolateWitnessTriggers()` localizes witness by taking increasingly smaller steps until a sufficiently-narrow step has been found during which the witness changes sign. (When the required isolation accuracy is strict, the root will be bracketed very tightly, such as between +1e-8 and -1e-8.) IsolateWitnessTriggers() then erroneously attempts to *repeat* the tiny step, assuming it will produce the identical behavior. With Jacobian re-use small, below-tolerance variations are expected and there is no reason to believe the function will change sign again.

### Proposed solution
As usual the goal of floating point computations should be to make discrete ("integer") decisions and never depend on sloppy numerical calculations for anything deterministic. The goal of IsolateWitnessTriggers() is to determine when an event has occurred, not to perform numerical gymnastics with touchy witness functions. When a sufficiently small step has isolated a witness zero to an interval, the event should be declared to have occurred in that interval. There is no further dependence on the witness function value.

The algorithm should be changed to be fully deterministic:
1. on entrance a large step from t0..t1 has been taken and one or more witnesses changes sign across the step
2. if h=t1-t0 is small enough, or we're not isolating at all, trigger the events and return
3. otherwise, use some method to estimate when the first zero crossing occurred, say tz
4. restart integration from t0..tz
5. if we now see no sign change, the trajectory has been advanced to tz with no events; just continue normally from t=tz -- sign changes may or may not recur later. Note that by taking a smaller t0..tz step we have a *more accurate* solution. The previous sign changes may have been artifacts.
6. if there is a sign change, let t1=tz and go back to step 2. 

Note that the main loop is driven by time -- it should not be a loop over witness functions as it is now.

sherm1, 2017-06-21:
When this issue is fixed, remove the PR #6265 workaround in simulator_test.cc marked with "TODO(sherm1).*6396".

edrumwri, 2017-06-25:
The analysis in the "cause" specified is mostly correct- incorrectness in the analysis lies in the statement that, "IsolateWitnessTriggers() then erroneously attempts to repeat the tiny step, assuming it will produce the identical behavior." This attempt to repeat the step is not erroneous but is necessary to ensure the correctness of the result- if the sign does not change as expected, the witness function is prone to triggering immediately over the _next_ integration interval. @sherm1's misunderstanding here indicates that the postcondition documentation for IsolateWitnessTriggers needs to be updated to reflect that the witness function(s) sign has changed at function termination.

I will instead attribute the cause to inability to repeat the computation in the implicit integrator when reuse is activated due to hidden state (Jacobian matrix and iteration matrix factorization) in the integrator. 

### Proposed solution

Integrators need the ability to repeat a computation such that an *identical result is obtained both times*. 

### But...

@sherm1's analysis was useful, still, by causing me to think about a witness function w1 that doesn't trigger over the full interval [t0, tf] but *does* trigger over the interval [t0, tc], where tc has been obtained by chopping the interval down to isolate the zero for a different witness function w0 (note that the opposite case, where the witness function initially triggers then no longer triggers as the interval is chopped down, has already been considered and is part of the current test suite). I will prepare a solution to this anticipated problem as well.

sherm1, 2017-06-25:
Nice to have you back, @edrumwri!

> Proposed solution: Integrators need the ability to repeat a computation such that an identical result is obtained both times. 

That's not possible in general, and witness functions must work in general. In the solution I proposed there is no need to repeat anything. Also, repeating an integration step can be very expensive so is a good thing to avoid if you can!

edrumwri, 2017-06-25:
Thanks Sherm!

My proposal is a pair of virtual functions IntegratorBase::SaveHiddenState() and IntegratorBase::LoadHiddenState(), that permit the computation to be deterministic (i.e., you should be able to guarantee that from the same hidden state and same context, the integration will produce the same result). For what class of integrator would that strategy fail?

Note that the code that repeats the integration is only a safety measure- as the documentation states, it can be removed once this invariant is safely satisfied.

sherm1, 2017-06-26:
That would fail for any integrator we didn't write, such as Radau5 or DASSL and means an extra Context-like architecture for our integrators to collect all internal state in one place. Could be done, but why? We don't need to depend on that.

edrumwri, 2017-09-22:
Addressed by #6434.

closed 2017-09-22 by edrumwri