RobotLocomotion/drake #6410: TypeSafeIndex comparison with size_t does not compile

amcastro-tri, 2017-06-22:
The following code compiles with a signed/unsigned comparison error (actually a warning then understood as an error).
```c++
using AIndex = TypeSafeIndex<class A>;
AIndex index;
std::vector<int> v;
v.push_back(1);
//EXPECT_TRUE(index == v.size()); // Compiler complains with warning (error).
// All other options below compile just fine.
EXPECT_TRUE(index == size_t(1));
EXPECT_TRUE((index == static_cast<unsigned long>(1)));
EXPECT_TRUE((index == static_cast<std::vector<int>::size_type>(1)));
unsigned long ul = 1;
EXPECT_TRUE(index == ul);
EXPECT_TRUE(ul == index);
EXPECT_EQ(index, 1);
```

SeanCurtis-TRI, 2017-06-22:
Stand-alone version which doesn't depend on drake to reproduce the problem:

```c++
// A test to reproduce the behavior we're getting in which comparison between
// our "type-safe index" and an unsigned int has inexplicable compiler warnings.
//
// Compile with:
//    clang++ --std=c++14 -Wsign-compare -Werror main.cpp

#include <iostream>
#include <vector>

class Index {
public:
  explicit Index(int i) : index_(i) {}
  operator int() const { return index_; }
  bool operator==(const Index& other) { return index_ == other.index_; }

private:
  friend std::ostream& operator<<(std::ostream&, const Index&);
  int index_;
};

std::ostream& operator<<(std::ostream& out, const Index& i) {
  out << "I(" <<  i.index_ << ")";
  return out;
}

#define COMPARE(expr) std::cout << #expr << ": " << (expr) << "\n";

int main() {
  const int val = 15;
  Index i(val);
  size_t sz = 3;
  std::cout << std::boolalpha;
  std::cout << "Test values:\n";
  std::cout << "i = " << i << "\n";
  std::cout << "sz: " << sz << "\n";

  // These test should all work
  std::cout << "\nWorking tests...\n";
  COMPARE(i == 0);
  COMPARE(i == val);
  COMPARE(i == size_t(val));
  COMPARE(i == static_cast<unsigned long>(val));
  COMPARE(i == static_cast<std::vector<int>::size_type>(val));
  COMPARE(i == 15u);
  COMPARE(i == static_cast<int>(sz));

  // These tests should all cause signed errors
  std::vector<int> values;
  for (int i = 0; i < val; ++i) values.push_back(i);
  //  COMPARE(i == sz);
  //  COMPARE(sz == i);
  //  COMPARE(i == values.size());
  return 0;
}
```

SeanCurtis-TRI, 2017-06-22:
The mystery deepens:

If you change this line:

```c++
const int val = 15;
```

to 

```c++
int val = 15;
```

Then *some* of the tests that build no longer build:

```c++
i == size_t(val);
i == static_cast<unsigned long>(val);
i == static_cast<std::vector<int>::size_type>(val);
```

So, const-ness matters.

SeanCurtis-TRI, 2017-06-22:
Without knowing the *why* of what is going on here. 2 monkeys at 2 keyboards have concluded that simply whitelisting integral types (`int`, `long`, `long long`, `unsigned`, `unsigned long`, `unsigned long long`) makes things happy (`const` or not).

jwnimmer-tri, 2017-06-22:
Why not just have the call site cast the other type (such as `size_t`_ to `int`, like we do everywhere else we mix `container::size()` results with other-code `int`s?

SeanCurtis-TRI, 2017-06-23:
That is, of course, what currently happens. This subject arises from the aesthetic desire to clean up the call site. And in the attempt to clean things up, that's where we discovered the surprise.

What's particularly interesting (what makes it an issue, particularly), is that the unit test does something that *claims* to suggest that the comparison is valid. So, at the very least, the test needs to be changed so what is implied is correct. The documentation should probably also be updated to account for this use case.

jwnimmer-tri, 2017-06-23:
I see.  I agree that the unit test should not test direct `OP`s with types other than `int`.

mwoehlke-kitware, 2017-06-23:
Why does `TypeSafeIndex` not use the actual index type of the things being indexed? That there seems like a bigger defect...

sherm1, 2017-06-23:
Per styleguide, we use signed ints wherever possible for indices. The quirks of the particular container being used internally are mostly not relevant in the API.

jwnimmer-tri, 2017-06-25:
> ... aesthetic desire to clean up the call site.

I would argue its a readability problem to have `TypeSafeIndex` magically inter-operate with `size_t` when regular `int` indices do not.  If there is a type conversion happening, that fact should be reflected at the call site.  If `static_cast<int>()` is too unsightly, then perhaps you should propose a `drake::as_int(size_t)` helper (or something) that everyone can use.

SeanCurtis-TRI, 2017-06-26:
Two points:

1. _Philosophical_: I'm 90% on board with that reasoning. It is *consistent*. But given both `size_t` and `int` can be indiscriminately used as index values in C++, it would make sense that the `TypeSafeIndex`, which is more a *semantic* entity than an *integral* entity, could arguably be implicitly interoperable with both. (That said, if we wanted to support that fully, it should probably contain a `size_t` instead of an `int` as that is the natural progression of integral promotion.)
2. _Practice_: Casting `size_t` to `int`s is what we do all over the code base. So, doing so in this particular context follows a very common and familiar visual paradigm. Based on that, killing ourselves to remove what could be cruft isn't worth it. 

Moral of the story: So, we'll close this after cleaning up the tests/docs.