RobotLocomotion/drake #6426: A g++-5.4.0 Issue with ASan

m-chaturvedi, 2017-06-26:
Consider:
`bazel test  --config=asan --compiler=gcc-5  //drake/automotive:automotive_simulator_test`

After updating bullet3, https://github.com/RobotLocomotion/drake/pull/6409, the above test fails.
This is an optimization bug in g++-5.4.0 which happens when `-O1` or `-O2` is used.  It does not happen when `-O0` is used.  Our Bazel's default is `-O1` for asan.

To reproduce with a 'simpler' example, compile the following code with, `ASAN_OPTIONS="check_initialization_order=true"` and `g++-5 -O1 -fsanitize=address`:
https://github.com/gcc-mirror/gcc/blob/gcc-6_3_0-release/gcc/testsuite/g%2B%2B.dg/asan/pr77396.C

This has been reported and supposedly fixed in g++-6.x.
Report and patch: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=77396
Builds failing: https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-bazel-nightly-memcheck-asan/

m-chaturvedi, 2017-06-26:
Checked with the latest tag for Bullet (2.86.1 ) at Jamie's suggestion.  The test passes.

jamiesnape, 2017-06-26:
Can you PR an update to that tag for both Bazel and CMake?

closed 2017-06-28 by jamiesnape