RobotLocomotion/drake #6460: Teach ZeroOrderHold to handle AbstractValue's

EricCousineau-TRI, 2017-06-28:
Per this [Slack discussion](https://drakedevelopers.slack.com/archives/C2CHRT98E/p1487355677000718), `ZeroOrderHold` only handles vector-valued inputs / outputs.

At present, it is possible to make a `AbstractValue`-compatible zero order hold, but (AFAIK), you must specify the type at the time of port / state declaration (given the present allocation semantics within the system framework).

The steps to achieve could be to:

- [ ] First create and test a `AbstractZeroOrderHold<ValueType>`, which will be a separate class that requires the input/output type.
- [ ] Permit declaration of completely type-erased ports and states at declaration within the system framework.
- [ ] Fold the functionality of `AbstractZeroOrderHold<>` into `ZeroOrderHold`, per Jeremy's suggestions on overloading options.

This is related to #6458.

\cc @kunimatsu-tri @jwnimmer-tri

EricCousineau-TRI, 2017-06-28:
For general syntactic sugar, it might(?) be useful to have some sort of diagram-wrapping mechanism that can (a) contain a direct-feedthrough system, (b) perform a consistent zero order hold on all outputs (similar to having an atomic subsystem in Simulink with a specified sample time), and (c) permit convenient mapping using the ports from the wrapped system.

jwnimmer-tri, 2017-06-28:
> (At present, type-erasure seems to only happen at the API-level.)

`LeafSystem::DeclareAbstractState` is http://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_leaf_system.html#a0f5b3aa23038e099188fdb6005c37c40

`LeafSystem::DeclareAbstractOutputPort` http://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_leaf_system.html#ab2d3c293055d499160857e5da0f2507c

What's missing?

EricCousineau-TRI, 2017-06-28:
> What's missing?

As I understood it, the APIs presently exposed still needs a derived class of `AbstractValue` upon allocation (e.g. `DeclareAbstractOutputPort(..., OutputType...)`), which at present would mean an instance of `Value<DataType>` (where `DataType` is the type for the port or state). Even if an abstract port / state is declared, I believe it still needs to be allocated. (I had looked at `SetDefaultState()`, but I assumed that this would require the abstract states to be allocated, and I wouldn't be able to use `SetDefaultState()` to query the upstream dummy value to populate the allocators.)

I had briefly tinkered with making a type of `DeferredAbstractValue` class that would permit instantiation without a concrete type attached to it (type erasure at declaration time), but ran into some snags on virtualizing downcasting to permit the ZOH output port (of type `DefferredAbstractValue`) to be compatible with downstream input ports of `Value<DataType>` - given that the present implementation of `AbstractValue` expects that the instance will be of type `Value<T>`:
https://github.com/RobotLocomotion/drake/blob/f5d188a/drake/systems/framework/value.h#L199

I'll double-check the API again; if there is an easy solution that I did miss, please let me know!

jwnimmer-tri, 2017-06-29:
The, e.g., `DeclareAbstractOutputPort` that I linked to does not require knowing `OutputType` at declaration; you just provide a functor that can produce one when needed.  Similarly, overriding `AllocateAbstractState` allows you to allocate a concrete value at runtime, not declaration-time.  That may not be good enough (e.g., where do we learn the passthrough type from?), but you definitely have the ability to provide a value later than declaration-time.

In any case a `ZeroOrderHold` ctor overload that takes an `AbstractValue` instead of `int size` should meet the need for now, without needing to make the passthrough type part of the class's type signature.  That can be done now, in a single PR, without the multi-step sequence shown above.

EricCousineau-TRI, 2017-06-29:
That sounds like a much simpler solution, thanks!
I will update the above steps, and then make a quick mention in #3109 to move the abstract value propagation there.

closed 2017-08-07 by EricCousineau-TRI