RobotLocomotion/drake #6490: Move non-vector portions of SisoVectorSystem into SisoSystem, have abstract-capable ZeroOrderHold leverage this

EricCousineau-TRI, 2017-06-30:
The present PR in flight (#6488) is teaching `ZeroOrderHold` to use `AbstractValue`s. As such, it needed to diverage from `SisoVectorSystem`, but still copy a portion of its API.

This could be fixed by adding a `SisoSystem`, which is type agnostic.

jwnimmer-tri, 2017-06-30:
I considered such a class when making the vector-specific variant, but decided there wasn't enough meat left nor use cases to make it worthwhile once the vector specialization stuff goes away.

EricCousineau-TRI, 2017-06-30:
I agree that there won't be much meat left (and the unittests should either be simple or next to non-existent), but I think the use cases are now there with `ZeroOrderHold` (and eventually, `PassThrough`).

jwnimmer-tri, 2017-06-30:
But would the ZOH and PassThrough implementations be more clear robust for having inherited from such a base?  I suspect not, but I guess the code will speak for itself.

EricCousineau-TRI, 2017-06-30:
I namely want de-duplicate the interface listed here (along with the related invariants in `AllocateContext`, if useful and not requiring obtuse `is_abstract()` logic) so it's a little easier for the API to evolve, if need be:
https://github.com/robotlocomotion/drake/blob/e307df9bd66f07ffbd4e589135d7e088047dc3be/drake/systems/primitives/zero_order_hold.h#L36
(not sure how long this SHA will stay alive - couldn't find a Review Number + Line Number permalink in Reviewable)

jwnimmer-tri, 2020-12-09:
I don't see anything remaining in `zero_order_hold.{h,cc}` that would be appropriate for extraction into a reusable base class.  Maybe #13898 / #13883 already resolved this?

EricCousineau-TRI, 2020-12-09:
Not sure if those resolved it, but I think this issue lost steam / value (saving code via inheritance ain't always the best choice).
Closing!

closed 2020-12-09 by EricCousineau-TRI