RobotLocomotion/drake #6500: pydrake_mathematical_program_ipopt_solver_test fails when NO_IPOPT=ON

RussTedrake, 2017-07-02:
my .bazelrc defines 
```
build --define=NO_IPOPT=ON
test --define=NO_IPOPT=ON
```
and I see
```
//drake/bindings:pydrake_mathematical_program_ipopt_solver_test          FAILED 
```

jwnimmer-tri, 2017-07-02:
Setting `--define=NO_IPOPT=ON` is currently not a supported configuration (and even moreso for running all tests).
- https://github.com/RobotLocomotion/drake/blob/60b5ed9724428e968a3b4029221cd1e5261af22a/tools/BUILD#L302-L309
- http://drake.mit.edu/bazel.html#proprietary-solvers

I'll see next week if there is a reasonable quick fix, otherwise will backlog this.

What's the reason for turning off IPOPT?  Perhaps its slow build?  We should be using a pre-compiled IPOPT anyway -- I'll open an issue on that next week.

RussTedrake, 2017-07-02:
not just a slow build, but ipopt was building everytime i flipped between console bazel commands and bazel commands from clion (some argument was different which retriggered the build every time).  agreed that precompiled ipopt should resolve that problem.

RussTedrake, 2017-07-02:
n.b. -- all other tests pass when this option is set.

PR #6512 (2017-07-08): Adjust pydrake unit tests for when IPOPT is disabled
Fixes #6500.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6512)
closed 2017-07-08 by soonho-tri