RobotLocomotion/drake #6513: Don't use genrule to compile IPOPT

jwnimmer-tri, 2017-07-05:
Using a `genrule()` to compile C++ software is frowned upon by Bazel upstream, but that's currently how we're building IPOPT.  We should either write `BUILD` file(s) for IPOPT, create a pre-compiled shared library release (e.g. with a Dockerfile), or use the OS's default build of IPOPT (debs /homebrew).  (The OS build may be insufficient; see #4913).

There are several problems with using a genrule:
- the compile step is slow (lots of software), and even with "half the cores" it takes a while locally and is worse in CI;
- because its a genrule, its dependencies are too wide (e.g. on environment variables) and thus it recompiles too often, such as when switching between CLion and console builds (see #6500);
- while we could pass `$COMPILATION_MODE` through the genrule (debug vs release), we cannot pass any other e.g. `$CXX` or `$CXXFLAGS` from `CROSSTOOL` into the compilation step.  Thus, our builds will _always_ use GCC for IPOPT, even if the user has otherwise selected `clang` as their compiler.

Probably the #4913 fix should be rolled into this effort.

jamiesnape, 2017-07-06:
I have some Fortran rules that really just need some cleanup, and then we could build natively with Bazel.

jwnimmer-tri, 2017-09-13:
See also #4913 where we are discussing what to do with IPOPT.

PR #7027 (2017-09-21): Use system IPOPT
This PR changes Bazel to seek IPOPT via pkg-config (i.e., from the host system) on Mac and Xenial, but leaves it unchanged on Trusty.  (The Trusty version of IPOPT is broken, so we should not use it.)  This unconditionally removes the install target for IPOPT, because we will not support installation on Trusty.

This makes the genrule build uncommon enough to declare this closes #6513.

This solves #4913 for Mac and Xenial, but I'll leave that issue open until Trusty is removed.

For now, we expect pre-merge CI to fail, because this adds new packages to install_prerequisites.  [... cut]
closed 2017-09-21 by jwnimmer-tri