RobotLocomotion/drake #6517: Orphaned MATLAB-related C/C++ and Java files

jamiesnape, 2017-07-06:
C/C++:

* [LCMScope.h](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/util/LCMScope.h)
* [makeFunction.h](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/util/makeFunction.h)
* [mexify.h](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/util/mexify.h)
* [standardMexConversions.h](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/util/standardMexConversions.h)

Java:

* [BotLCMGLClient.java](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/util/BotLCMGLClient.java)
* [CoordinateFrameData.java](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/util/CoordinateFrameData.java)
* [MyLCMTypeDatabase.java](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/util/MyLCMTypeDatabase.java)
* [LCMCoder.java](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/util/LCMCoder.java)
* [MessageMonitor.java](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/util/MessageMonitor.java)
* [Transform.java](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/util/Transform.java)

Should they be compiled into something or removed?

jamiesnape, 2017-07-06:
And just to double check, it is correct that we are still using [barycentricInterpolation.cpp](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/util/barycentricInterpolation.cpp), [lcmLogger.cpp](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/util/lcmLogger.cpp),  and [realtime.cpp](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/util/realtime.cpp)?

RussTedrake, 2017-07-20:
not using any of those three anymore.

jamiesnape, 2017-07-20:
Can anything in the top list be removed?

RussTedrake, 2017-07-20:
the mexify tools are a strong candidate for writing our matlab bindings when we start using them again.  But given the matlab code has been zapped, it is not being used now.   i think it's aok to zap everything on the top list as well.   cc @jwnimmer-tri

jwnimmer-tri, 2017-07-20:
SGTM.  git remembers.

PR #6611 (2017-07-20): Zap orphaned and unused MATLAB-related C++ and Java
Closes #3982, closes #6517.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6611)
commit 47c4c56d (2017-07-20): Zap orphaned and unused MATLAB-related C++ and Java

Closes #3982. Closes #6517.
closed 2017-07-20 by soonho-tri