RobotLocomotion/drake #6518: Update drakeURDF.xsd

SeanCurtis-TRI, 2017-07-06:
The XML Schema Definition (xsd) file in which we elaborate Drake's support for URDF features (particularly with regards to Drake-specific features) has fallen out of sync with the current code/functionality. It needs to be triaged and updated so that Drake's functional state is reflected in the document.

This arose in a discussion of a PR extending this functionality with a ball joint: https://reviewable.io/reviews/robotlocomotion/drake/6489#-KoNVLpgYWG4FywxzM6q:-KoNcONwt2sybgUsB0bo:b-d2fwnl

EricCousineau-TRI, 2019-06-24:
This has cropped up in #11589. Should we upgrade priority, as the file seems to be more of a distraction than anything?

SeanCurtis-TRI, 2019-06-24:
Do we actually validate against it? Do we plan on validating against it? What's the cost of maintaining it vs the value of having it up to date?

EricCousineau-TRI, 2019-06-24:
I don't think so. I'm OK with "delete this file" as being *part* of the solution; that's better than it being a lie.

However, at some point, we *should* introduce something that defines what modifications we have on top of URDF / SDFormat that are actively being used, and should validate against it. However, that can be split off as a separate issue.

Does that sound good to you? If so, I'll create the issue for it.

SeanCurtis-TRI, 2020-05-14:
A year later, I'm coming back to this.

1. I think #11589 was the wrong link.
2. Between deleting the lie and keeping the lie, deleting is definitely preferred.
3. Between deleting the lie and stoppping the lie, stopping is definitely preferred.

EricCousineau-TRI, 2020-05-14:
> I think #11589 was the wrong link.

I mean, MOSEK relates to everything, no? :P (sorry about that!)

And yeah, ideal is stopping the lie. However, I'm up for deleting the lie for short term!

SeanCurtis-TRI, 2020-05-14:
1-2-3 NOT IT!  :finger_on_nose:

EricCousineau-TRI, 2020-05-14:
As far as the "split off on separate issue", I think that is now covered by https://github.com/RobotLocomotion/drake/issues/13103 - it's already on me, so I'm good there.

Would you be able to take on deleting the file? (I know this is against the rules... but I play dirty!)

SeanCurtis-TRI, 2020-05-14:
Yeah. I think I'm just about up to the task.

PR #13326 (2020-05-18): [doc] Remove the incorrect drakeURDF.[xpr|xsd]
The schema defined in drakeURDF.xsd is significantly out of date. We don't have the time or resources to bring it up to date. For now, we'll just remove it so as to avoid the suggestion that it is reliable. It can always be resurrected from git when it becomes a priority.

This also removes references to the file.

resolves #6518

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13326)
closed 2020-05-18 by sammy-tri