RobotLocomotion/drake #6523: Bring back the littledog URDF

hongkai-dai, 2017-07-07:
After removing the submodule littledog from drake repo, we cannot create a little dog model any more. Is it OK that we bring back the URDF of little dog back to `examples` folder? 
@jwnimmer-tri @RussTedrake

RussTedrake, 2017-07-07:
Short answer: yes from me.
Slightly longer version:  we resisted putting it in originally because we wanted to avoid bloating git history with mesh files.  This (and #6417) might be good first candidates for pulling them in from a separate source via bazel, via #6124 or some alternative?

jwnimmer-tri, 2017-07-07:
(For the record: removed in #6231.)

Yes, of course if we need the model data then we should PR it back, no problem.  Is the plan to write some sort of demo in Drake's tree that relies on the model?

However, we shouldn't simply revert #6231, because it used git submodules.  Since the data is in git already, adding `github_archive` to the `WORKSPACE` (plus a `tools/littledog.BUILD` file that `filegroup`s the model data) should be trivial and meet the need.  Later we could refactor to use #6124 if desired.

jwnimmer-tri, 2017-07-07:
> Is the plan to write some sort of demo in Drake's tree that relies on the model?

FYI -- I ask in particular because if the only goal is "put LittleDog URDF back on the internet" then that's already done, because it's on master in a separate repository.  If for some reason Drake needs to compute using the model, then of course yes we can bring it back as a dependency.

hongkai-dai, 2017-07-07:
> Is the plan to write some sort of demo in Drake's tree that relies on the model?

Exactly. I plan to write some IK tests on Little Dog, to test the IK capability on a floating base robot. The test is for both snopt IK and global IK, I plan to start with little dog instead of Atlas or Valkyrie, since little dog has fewer degrees of freedom.

jwnimmer-tri, 2019-09-19:
@hongkai-dai Do you still plan to test IK using LittleDog?  Maybe the issue should be titled as "Add IK tests" not "Add some URDF file" (assuming there is no such issue already open).

hongkai-dai, 2019-09-20:
I don't think I will add the IK test to LittleDog. I would like to keep the name of the issue as adding little dog URDF.

jwnimmer-tri, 2019-10-10:
The URDF is available at https://github.com/RobotLocomotion/LittleDog.  Once we have a motivating case to add it to the Drake WORKSPACE file, we can easily do that as needed.  I don't see a need for a tracking issue for that.  Please re-open if I've misunderstood.

closed 2019-10-10 by jwnimmer-tri