RobotLocomotion/drake #6527: Investigate replacing the joint friction model and/or implementation under RigidBodyTree

edrumwri, 2017-07-07:
The current single DOF joint friction model in RigidBodyTree (under multibody/joints/fixed_axis_one_dof_joint.h),

f = -μᶜ sgn(q̇) - μᵛ q̇

where the first term is supposed to represent Coulomb friction and the second represents viscous friction, is one that is commonly employed in robotics applications. However, we lack knowledge of  whether and how this model has been validated. The Coulomb friction aspect is also simplistic- it does not account for the constraint force at the joint- and does not model stiction.

The current implementation in Drake uses a parameter, w, whose default value causes a divide-by-zero error:

f = -μ* sgn(q̇) - μᵛ q̇      
where
u* = min(1, max(-1, q̇/w))⋅μᶜ

No documentation exists for this aspect of the implementation, meaning that its purpose and limitations are generally unknown.

RussTedrake, 2017-07-12:
The documentation for the coulomb_window (and all other drake-specific tags) is here:
http://drake.mit.edu/urdf/drakeURDF.html

and the default in the matlab parser was indeed eps:
https://github.com/RobotLocomotion/drake/blob/last_sha_with_original_matlab/drake/matlab/systems/plants/RigidBody.m#L29

RussTedrake, 2017-07-12:
And note that the existing model, or similar piecewise-affine approximations of coulomb friction, may still have a lot of value for control design.

edrumwri, 2018-10-08:
RBT's limited remaining life makes this RBT-specific issue nearly moot. @amcastro-tri might want to consider this for MBT.

closed 2018-10-08 by edrumwri