RobotLocomotion/drake #6544: Ensure behavior of -Werror=shadow is consistent between Bazel and CMake

EricCousineau-TRI, 2017-07-10:
I had attempted to address this in #6543 by adding `-Werror=shadow` to the bazel build, but was unable to reproduce the error in CI (mentioned in #6540) on my system using the changes in #6543 and the original error from #6488.

jwnimmer-tri, 2017-07-10:
Did you test on OS X clang?  I think the problem was OS X specific.

Since turning off CMake build is imminent, probably this issue is more appropriately titled "figure out why Bazel builds didn't complain about the issue that #6540 fixed, and then if the answer was that they _should_ have complained, fix the build system until they do".

EricCousineau-TRI, 2017-07-10:
I had tested this on a Mac 10.12 system with `clang`, confirming that I saw `-Werror=shadow` in the output (using `bazel ... -s`), but did encounter the error.

Will try to reproduce this outside of `drake` before attempting to fix this here. (Will also change the title.)

EricCousineau-TRI, 2017-07-31:
Following this build log using AppleClang 7.3.0:
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-clang-continuous-debug/2535/consoleText

It seems like that `-Werror=shadow` will cause a similar error with GCC, but not with newer versions of clang (tested on AppleClang 8.1.0):
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=57709

This did not appear with Bazel on GCC because the flags were not present. I will submit a PR to add this in.
For Clang, it still remains to be seen if there is flag that is as strict as GCC's `-Werror=shadow`.

See these reproduction steps:
https://github.com/EricCousineau-TRI/repro/blob/18ae4ca11d236cc5fa4f0ba9618abac2573300c7/cpp/flags/shadow/Makefile

EricCousineau-TRI, 2017-08-18:
Resolved by #6721.

closed 2017-08-18 by EricCousineau-TRI