RobotLocomotion/drake #6552: Promote ASan, TSan, and Valgrind builds to Nightly Production?

jamiesnape, 2017-07-12:
With the exception of the timeout of `//drake/common:symbolic_polynomial_test` that was introduced last night (which @m-chaturvedi is currently fixing), these jobs have all been clean for several days and I would like to promote them to be monitored by build cops so that they stay that way.

* [linux-xenial-clang-bazel-nightly-memcheck-asan](https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-bazel-nightly-memcheck-asan/)
* [linux-xenial-clang-bazel-nightly-memcheck-asan-everything](https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-bazel-nightly-memcheck-asan-everything/)
* [linux-xenial-clang-bazel-nightly-memcheck-tsan](https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-bazel-nightly-memcheck-tsan/)
* [linux-xenial-clang-bazel-nightly-memcheck-tsan-everything](https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-bazel-nightly-memcheck-tsan-everything/)
* [linux-xenial-clang-bazel-nightly-memcheck-valgrind](https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-bazel-nightly-memcheck-valgrind/)
* [linux-xenial-clang-bazel-nightly-memcheck-valgrind-everything](https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-bazel-nightly-memcheck-valgrind-everything/)
* [linux-xenial-clang-nightly-memcheck-asan](https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-nightly-memcheck-asan/)
* [linux-xenial-clang-nightly-memcheck-tsan](https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-nightly-memcheck-tsan/)
* [linux-xenial-clang-nightly-memcheck-valgrind](https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-nightly-memcheck-valgrind/)
* [linux-xenial-clang-nightly-memcheck-valgrind-minimal](https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-nightly-memcheck-valgrind-minimal/)
* [linux-xenial-gcc-bazel-nightly-memcheck-asan](https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-bazel-nightly-memcheck-asan/)
* [linux-xenial-gcc-bazel-nightly-memcheck-asan-everything](https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-bazel-nightly-memcheck-asan-everything/)
* [linux-xenial-gcc-bazel-nightly-memcheck-tsan](https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-bazel-nightly-memcheck-tsan/)
* [linux-xenial-gcc-bazel-nightly-memcheck-tsan-everything](https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-bazel-nightly-memcheck-tsan-everything/)
* [linux-xenial-gcc-bazel-nightly-memcheck-valgrind](https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-bazel-nightly-memcheck-valgrind/)
* [linux-xenial-gcc-bazel-nightly-memcheck-valgrind-everything](https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-bazel-nightly-memcheck-valgrind-everything/)
* [linux-xenial-gcc-nightly-memcheck-valgrind](https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-nightly-memcheck-valgrind/)
* [linux-xenial-gcc-nightly-memcheck-valgrind-minimal](https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-nightly-memcheck-valgrind-minimal/)

FYI @jwnimmer-tri @stonier.

jwnimmer-tri, 2017-07-12:
SGTM.

jamiesnape, 2017-07-12:
Done.

closed 2017-07-12 by jamiesnape