RobotLocomotion/drake #6575: Issue Building Drake with ROS

cirpote, 2017-07-14:
Ciao,

I am trying to build drake with ROS Kinetic on my laptop (with Ubuntu 16.04) following your guide, [Building Drake as a Catkin Package in ROS](http://drake.mit.edu/from_source_ros_kinetic.html) but i get this error:

[image]

This error happens after the first tentative of building since there is no VTK5 installed on my laptop and i need to build it directly from source. Following your guide i to set -DUSE_SYSTEM_VTK=OFF -DWITH_VTK=ON, but actually both flags do not exist and are not settable.

Recompiling i get again the above error.

liangfok, 2017-07-14:
Hi @cirpote, which commit SHA are you using? I just want to confirm you're using the latest version of Drake. You can view the SHA by executing:

```
git log -1 --format="%H"
```

cirpote, 2017-07-14:
I am using this version:

80b0dfeff71e3681d26e48a97da6f762fece00c9

liangfok, 2017-07-14:
I'll try to replicate the error later this morning. I suspect #6476 may be related.

jamiesnape, 2017-07-14:
You would need to switch `WITH_DIRECTOR=OFF`.

cirpote, 2017-07-14:
Now it's working! thanks a lot

PR #6580 (2017-07-17): Updates Drake's ROS Kinetic build instructions to disable Director
Resolves #6575.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6580)
closed 2017-07-17 by jwnimmer-tri