RobotLocomotion/drake #6581: Add path of drake

yaohuig, 2017-07-16:
Is there any instruction on how to add path of drake under bazel and cmake build? Thanks!

liangfok, 2017-07-16:
I'm not sure if this is what you're asking about but Drake checks an environment variable called `DRAKE_RESOURCE_ROOT` that you could set to point to `drake-distro/`.

https://github.com/RobotLocomotion/drake/blob/60e67269173846fe258c18a7e8702012266e8d9b/drake/common/find_resource.cc#L163-L164

yaohuig, 2017-07-16:
Thanks for the reply. I want to include the Drake library in my project, so what command should I use to add its path? Thanks.

liangfok, 2017-07-16:
I believe the following repository is still under construction, but maybe there's enough there to help you get unblocked?

https://github.com/RobotLocomotion/drake-shambhala

yaohuig, 2017-07-16:
The code in this repository needs to include each .h file and add dependency manually. Is there any easier way to add the dependency so I can use the API of Drake in my project? Thanks!

jwnimmer-tri, 2017-07-16:
@liangfok The master branch of shambhala is not appropriate to point users to.  The PR 8 in shambhala is more like what a better answer will be, though its still under review and refinement.

stonier, 2017-07-26:
@yaohuig please follow the development in [drake-shambhala](https://github.com/RobotLocomotion/drake-shambhala), in particular `drake_cmake_installed`. If you still have queries, please ask there and provide some context about your use case when you do.

closed 2017-07-26 by stonier