RobotLocomotion/drake #6603: kcov is not working in Ubuntu

soonho-tri, 2017-07-19:
Recently @hongkai-dai and I found that `bazel test --config=kcov` is not working in Ubuntu. Here is a screenshot showing its symptom.

[image]

We've tried different versions of `kcov` including `v32`, `v33`, `v34`, and the `master` of https://github.com/SimonKagstrom/kcov. None of them is working in Ubuntu, which makes us suspect that it's the new version of `bazel` which causes this problem. 

Interestingly, in OSX, bazel kcov is working as expected (for example, `bazel-0.5.2-homebrew` + `kcov v32-6-g045d4`).

rpoyner-tri, 2017-07-24:
The symptom is a result of changes in bazel itself, between release versions 0.4.5 (works) and 0.5.0 (kcov broken). Specifically, commit f396da66ac2421373172e88538066051a22f0309 in the bazel source repo is where things stop working.

Don't know why yet, but at least we know when and where.

rpoyner-tri, 2017-07-24:
Roughly speaking, `kcov` cannot find the project source code under the new bazel execution environment. Without the source code, it can't build its coverage report.

There is some wiggle room to finesse the file finding; haven't quite found the right solution yet.

rpoyner-tri, 2017-08-21:
Closing; fixed by https://github.com/RobotLocomotion/drake/pull/6880 .

closed 2017-08-21 by rpoyner-tri