RobotLocomotion/drake #6608: Python errors with Homebrew

m-chaturvedi, 2017-07-20:
We see `Fatal Python error: PyThreadState_Get: no current thread` errors when some Python tests are [run in director](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-clang-continuous-debug/2458/consoleText)

And @jamiesnape pointed out that they are perhaps because of this [change in Homebrew](https://github.com/Homebrew/homebrew-core/commit/e7f4e1c03abbe0842b77e3d163e737a81ca955e7#diff-313a6a3d25aa041ee3dc3cbd65d4f22b)

jamiesnape, 2017-07-20:
Note also that adding `/usr/local/opt/python/libexec/bin` does not fix the problem.

jamiesnape, 2017-07-20:
FYI @stonier. As mentioned briefly in the meeting earlier. Also affects Spartan.

patmarion, 2017-07-20:
I looked through the build log but couldn't find anything obvious, it does look like everything that is compiled is using the correct version of Python.  Is it possible that the VTK bottles are out of date or using the wrong version of Python?

I just started a Travis-CI test of director/master, since it hasn't run recently.  It should do a `brew update` so maybe the issue will show up there.

Is it possible to get SSH access to a machine where this issue is occurring?  Without interactive debugging, I think running `otool -L *.dylib` at the end of the build would be helpful information to see in the log.

jamiesnape, 2017-07-20:
The bottles were updated yesterday, so no.

EricCousineau-TRI, 2018-12-06:
Unsure of what the current state of this is; closing for now.

closed 2018-12-06 by EricCousineau-TRI