RobotLocomotion/drake #6619: AcrobotPlant unit testing

jwnimmer-tri, 2017-07-20:
As shown by the fix in #6599, Drake's `AcrobotPlant` lacks unit tests.

At a minimum, we should have test coverage for that one specific bug fix, but in general I think it would benefit Drake to have our key examples such as Acrobot demonstrate our best ideas on how to write and test Systems, and as such it should have more complete functional testing coverage.

(Note that the easiest way to ensure that #6599 has appropriate test coverage is likely by _removing_ code -- by changing the member fields into declared framework parameters per the existing TODO, we would be leveraging the framework's infrastructure and not need any Acrobot-specific paramters-survive-transmogrification test case!)

RussTedrake, 2017-07-21:
there are a number of algorithms that can be ported for Acrobot, especially now that we have SOS support in c++.  i would want to survey those before removing parameters/features.

jwnimmer-tri, 2017-07-21:
Heh, I was suggesting _adding_ features by _removing_ code.  Right now, the AcrobotPlant's parameters are direct member fields.  If they were System-declared parameters then they would be open to analysis (and require less custom code than they do now).

jwnimmer-tri, 2019-03-26:
I will figure out how to deduplicate this issue with the other related linked issues upthread.

jwnimmer-tri, 2019-05-01:
> Right now, the AcrobotPlant's parameters are direct member fields. If they were System-declared parameters then they would be open to analysis (and require less custom code than they do now).

FYI #8109 has already implemented the change to parameters handling.

(The lack of unit testing remains, so this issue is still open.)

jwnimmer-tri, 2020-09-28:
Re-assigning to @mitiguy to resolve.  In #13593 we're apparently adding yet another acrobot, buried inline in a unit test.  That's beyond what I can retrofit.

jwnimmer-tri, 2020-09-29:
In b43bf44d7148ad4a8ab919ddcc502aa1168b08a4 we changed parameters to the framework, so the particular regression bug that lacked a unit test is now moot -- we can rely on the framework to handle it properly.  There might be additional missing tests, but we'll only add them as we find _new_ bugs.

closed 2020-09-29 by jwnimmer-tri