RobotLocomotion/drake #6631: drake::is_numeric is misnamed

sherm1, 2017-07-21:
Drake's `common/number_traits.h` defines a `is_numeric<T>` but has to define `std::complex` as "non-numeric". Since that class represents complex *numbers* that is an odd classification! Per [this comment](https://reviewable.io/reviews/robotlocomotion/drake/6626#-Kp_wJRiG37Pj6k9_ba5), the classification is really intended to be something like `is_on_real_line<T>`, probably because it is used to determine whether these can be compared with `<` and `>`.

The name should be changed to reflect the intent.

DamrongGuoy, 2018-08-29:
I'd like to work on #6631 as a warm-up project in my second week.  We were discussing renaming `is_numeric<T>`, which determines whether T supports comparison with `<` and `>`. For example,
`is_numeric<double>` is true.
`is_numeric<std::complex>` is false.
`is_numeric<symbolic::Expression>` is false.

One candidate is `is_on_real_line<T>`, and I also think about `is_comparable_numeric<T>`.  Please let me know your preference, @sherm1 and @soonho-tri.  The third candidate is welcome too.  I will also work on deprecation, so it won't break other projects.

SeanCurtis-TRI, 2018-08-29:
Possibly also something along the ideas that it is "orderable"?

sherm1, 2018-08-29:
Thanks for doing this, @DamrongGuoy. 

How about `is_ordered_numeric<T>` or maybe better, `is_ordered_scalar<T>`? (Because we call our ubiquitous template type a "scalar", borrowing from Eigen's term.) I don't think `comparable` is quite right because it includes comparisons for equality, which std::complex, symbolic, etc. do support.

sherm1, 2018-08-29:
(Our comments crossed, @SeanCurtis-TRI but we were thinking the same thing.)

SeanCurtis-TRI, 2018-08-29:
Yeah....but github gives me credit for it first!  Bwa, ha, ha, ha!

DamrongGuoy, 2018-08-29:
`is_ordered_scalar<T>` sounds good to me!   I like the word "ordered."  It reminds me the way C++ STL distinguishes std::map v.s. std::unordered_map.

jwnimmer-tri, 2018-08-29:
I disagree that the trait being modeled is primarily to do with ordering.  The main question is whether or not the scalar can be mapped onto the real line in a reasonable way.

jwnimmer-tri, 2018-08-29:
In other words, I would propose `is_real_numberish` except it sounds lame.  But that's more the direction you want to go.  It's about numbers, not abound total / strong / weak / partial ordering etc. stuff.

jwnimmer-tri, 2018-08-29:
(Or maybe, since this trait is used in only one dark corner of Drake in the first place, having a lame name is a feature, not a bug.)

sherm1, 2018-08-29:
I'm not sure why you don't think this is about ordering, Jeremy, can you clarify? To both Sean and me (independently) that seemed like the key property. Usage in Drake is either
- a guard against doing a test to see if some value is "above" a tolerance, or
- whether it makes sense to check for one time being "later" than another.

Those are both ordering questions. It's not clear to me what other property would distinguish a real number from a complex one -- they support all the same operations except for ordering. In fact, `std::complex` can be used in place of AutoDiffScalar to calculate derivatives if there are no if statements, and complex augmented with ordering on the real part (sometimes called _surreal_ numbers) is a complete AutoDiffScalar replacement.

jwnimmer-tri, 2018-08-29:
So the easy answer is that when I went looking for uses of `is_numeric`, somehow I only found the ExtractDoubleOrThrow use.  I see the other ones now.

I think things like "Find the next update" by comparing times are actually defective code.  You should be able to calculate a System's event computations with symbolics, as long as the event times are literals (reals) at runtime -- the state or inputs could still be symbolic variables.  Those methods should ExtractDoubleOrThrow on the time values, and not have trait guards.

And the other uses like "is physically valid" enable_if's are really only asking if the type is double-convertible (so they can use Eigen's built-in computations for SVD, determinants etc.)  They do not seem to be sensibly conditioned on "are the types orderable", but rather just want "please disable me when symbolic expressions are in use".

sherm1, 2018-08-29:
I agree that some of those uses are sketchy. But I think it would be fine to define a trait "is_orderable_scalar" and implement it based on that property. It is mathematically sound and easy to explain, which is a definite improvement over the inaccurate `is_numeric` we have now. We can argue separately about the appropriate use of that trait or maybe define another trait like `is_convertible_to_double` for cases where that's really what we want to know. Would the latter actually be what we have been meaning by `is_numeric`?

jwnimmer-tri, 2018-08-29:
My thesis is that _every_ current use of `is_numeric` in Drake either (1) should be removed or (2) is intended to denote `is_always_convertible_to_double`.

I don't object in principle to adding `is_orderable_scalar` to Drake, but I do object to merely renaming `is_numeric` to be that, since I did not see any current uses of if whose guard _should_ be `is_orderable_scalar`.

sherm1, 2018-08-29:
OK, that makes sense to me. How about we rename `is_numeric` to `is_always_convertible_to_double` and check usages to make sure that makes sense everywhere? I would rather not remove any of the existing code at the same time but instead file an issue to change bad uses in a subsequent PR.

Any objections to that idea?

jwnimmer-tri, 2018-08-29:
Definitely agree with an approach to rename-first, then rework any misuses later.  (Or at least, to have them in separate PRs, whichever order they occur in.)

Even though I coined it, I still don't love `is_always_convertible_to_double`.  I am almost wondering if we just want "is_symbolic" and don't try to overgeneralize.  Give me a little more time to think about it.

jwnimmer-tri, 2018-08-29:
The challenge is that, for each current use of `is_numeric`, the _reason_ we need to guard the method is bespokely unique (and also not well-documented with a rationale).  The only thing they share in common is that they don't (yet) compile for symbolic.  Sometimes it's because SVD isn't supported.  Sometimes it's because we need to uniquely identify some non-numeric metadata associated with the minimal value in a set.  Sometimes it's because we've decided that we don't _want_ to enforce matrix validity checks for our symbolic scalar matrices, even though we could just as easily have done so.

I can see the appeal of having "Does scalar X support operation Y" traits so that downstream users (or features in Drake's future) can supply interesting scalars and have reasonable default things happen for a method that needs operation Y, I don't think we have yet demonstrated that we would be competent at flagging such traits successfully, and in any case I would be skeptical to try without testing -- which means either incorporating more of those scalars into our tests, or having "canary scalars" that we use in their stead.  The feature gaps between "double" and "my custom scalar X" are multi-dimensional, and "can convert to double" is but one of them -- and in much of our code not (as it happens) the one we truly care about, if we are being precise.  The "is_symbolic" is an admission of defeat and retreat to simplicity -- we have X = symbolic, we know this method doesn't support it, so we ad-hoc turn if off.  If later we can generalize to a more specific trait that was more precise, we can rework it at that point.

That all said, I don't know that `is_symbolic` is my vote in any case.  I am starting to think that the right answer here is to meet and argue over code, not ideas.  Possibly PR draft(s) where every current use of is_numeric is reworked, and addressed on an individual basis, is better?

I would be happy to start, myself, by fixing the "event time" uses, since it should be very quick, and would remove some traits checks entirely.

jwnimmer-tri, 2018-08-29:
(Or @DamrongGuoy could take a crack at the event time cleanup?  It should be approximately as simple as removing the `is_numeric` guard and then changing `if (timeA < timeB)` to  `if (ExtractBoolOrThrow(timeA < timeB))`, with suitable tests.)

jwnimmer-tri, 2018-08-29:
(I take it back -- we don't need an `ExtractBoolOrThrow`.  The only fix needed is a `ExtractDoubleOrThrow` on the `ceil(offset_time / period)))` expression.)

jwnimmer-tri, 2018-08-29:
> ... except for the loss of the int64_t conversion (unless I'm staring at it and missing it). That's there to kill off roundoff glitches which otherwise plague that algorithm ...

In my branch, the double division is still fed into `ceil`.  To the best of my knowledge, for `double` variables `a` and `b`, these two formulations are the same:
1. `a * ceil(b / a)`
2. `a * static_cast<int64_t>(ceil(b / a))`

Are there relevant cases where they are not the same?  My understanding is that `ceil` always returns an integer -- but still typed as a `double`.  I thought the only way they would not be the same was if the division was not finite, or if it overflowed `int64_t`.

> I'm not clear about the urgency of making that change, though -- the current compile-time guard would make sense if renamed is_ordered_scalar or is_always_convertible_to_double wouldn't it? Or are you trying to support the case where T=symbolic but the particular expression can be evaluated to a double?

I think the guard makes absolutely no sense.  Yes, when T=Expression we should do the full (real) logic, and only throw if `time` has unbound variables.  My branch already does that (I think; it compiles, at least).  All of the implicit Expression defaults just do the right thing magically.

The reason I propose doing this change (and then other similar ones) prior to the rename, is because I think that once we do them, there will be no more uses of `is_numeric` left in the tree, so we won't have to decide what it means.

I think its hard to come up with a better name to "just rename to temporarily for now".  Any rename induces compatibility hassles with deprecation, and if its not the right long-term name, we'll just have to keep renaming it again.  I think its hard to choose the right name without first agreeing on what semantics we want, and to agree on what semantics we want, I'd like to clarify all of the existing uses to be well-justified and clear on their purpose.

> ...  Just a summary for my own understanding ...

It is hard for me to say whether or not the table is correct, because Drake's current uses of `is_numeric` are inconsistent enough that I don't think we actually have any consistent semantics for what it means.

In short -- I'd like to propose that the next task here is to re-evaluate all uses of `is_numeric` guards in the tree, and try to remove as many of them as possible.

DamrongGuoy, 2018-08-30:
Thank you, @jwnimmer-tri.  I pretended to ask CLion to refactor/rename `is_numeric`.  There are three definitions (default `is_numeric`, `is_numeric<symbolic::Expression>`, and `is_numeric<NonNumericScalar>`).  There are 22 usages in 8 files among extract_double, rotation_matrix, multibody_tree, and framework.  Tomorrow (Thu 8/30/18) @sherm1 and I will go through them together.

jwnimmer-tri, 2018-08-30:
I have a new working branch with all of my proposals so far: https://github.com/jwnimmer-tri/drake/commits/kill-is_numeric2.

So far, I believe is_numeric will be entirely removed, with one of two options depending on circumstance:
1. No template guard needed (as per #9331, and then a few other future commits on my working branch).
2. Guard using the new `Bool<T>::is_native`.  The cases that _do_ need a guard all seem to be places where the question is whether or not formulas over `T` are a native `bool` or not (i.e., they can be used in C(++) `if` branches).  We already have a great type to characterize the properties of formulas over T scalars -- is called `Bool<T>` and we should use it, not reimplement it as `is_ordered_scalar`.

sherm1, 2018-08-30:
Sounds great, Jeremy -- would be great to see these evaporate. Then the present issue can be resolved by removing is_numeric and its header file. We'll hold off to see how this comes out.

PR #9453 (2018-09-14): common: Deprecate is_numeric and Bool<T>
Closes #6631.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9453)
closed 2018-09-14 by jwnimmer-tri