RobotLocomotion/drake #6638: Segmentation Fault upon launching drake-visualizer on OS X El Capitan

zamzami, 2017-07-23:
I have a relatively outdated version Drake which works perfectly fine on Matlab. Unfortunately, I have accidentally installed Gazebo 8 which messed up my Qt4 files and installed Qt5, which in turn gave the same error as in issue #5927. 

I tried to update the latest patch of VTK5    
`brew remove vtk5 && brew untap robotlocomotion/director && brew tap robotlocomotion/director && brew install vtk5`
 then rebuilt my outdated version of director but it failed. 

So I downloaded the latest version of director `SHA  aefc063` placed it in externals and built it successfully. However, it gives segmentation fault error upon launching  director 

```
 Last login: Sun Jul 23 18:29:38 on ttys000
/Users/ZZ/drakegit/drake-distro/externals/director/build/install/bin/drake-visualizer ; exit;
ziads-mbp:~ ZZ$ /Users/ZZ/drakegit/drake-distro/externals/director/build/install/bin/drake-visualizer ; exit;
Segmentation fault: 11
logout
```
I would also like to point out that importing vtk gave the same error

```
ziads-mbp:~ ZZ$ python
Python 2.7.10 (default, Aug 22 2015, 20:33:39) 
[GCC 4.2.1 Compatible Apple LLVM 7.0.0 (clang-700.0.59.1)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import vtk 
Segmentation fault: 11
ziads-mbp:~ ZZ$ which python2.7
/usr/local/bin/python2.7
ziads-mbp:~ ZZ$ python2.7 
Python 2.7.13 (default, Jul 22 2017, 23:50:44) 
[GCC 4.2.1 Compatible Apple LLVM 7.3.0 (clang-703.0.31)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import vtk 
Segmentation fault: 11
```
The issue could be related to some python path problem so here are my paths. (Homebrew doesn't fiddle with OS X system's python path anymore while python2.7 and python3 coexist in `/usr/local/bin/`
```
ziads-mbp:~ ZZ$ which python 
/usr/bin/python
ziads-mbp:~ ZZ$ which python2.7
/usr/local/bin/python2.7
ziads-mbp:~ ZZ$ echo $PYTHONPATH
/usr/local/opt/vtk5/lib/python2.7/site-packages:
ziads-mbp:~ ZZ$ cat /etc/paths
/usr/local/bin
/usr/bin
/bin
/usr/sbin
/sbin
```
Any help is highly appreciated :)

zamzami, 2017-07-24:
@patmarion I'm bringing this to your attention if you have time to look into it or if you can refer someone else :)

patmarion, 2017-07-24:
You can download the most recent pre-compiled version of Director for mac here:

https://bintray.com/patmarion/director/director/0.1.0-146-g4fd4613#files

It looks like your segfault is related to mismatched homebrew versions.  If you want to compile Director from source, perhaps you should try configuring it with `USE_SYSTEM_VTK=OFF` so that it will compile VTK from source rather than use the homebrew version.

zamzami, 2017-07-24:
Works like a charm. Thanks

closed 2017-07-24 by zamzami